borglab/gtsam #2: strange optimization behavior on node/factor removal on tracking and filtering application

emil-jose, 2018-05-28:
In order to avoid the ever growing factor graph problem, the oldest factors and nodes were removed from the graph and the remaining graph was given for optimization. The optimization works fine and outputs a filtered  result till the node with ID - half of the latest id in the graph is removed from the graph. Once the node with id half of that of the latest node is removed, the co-variance matrix of the latest node will go to a high negative value(-10^12) and the nodes added incrementally after that will hold this same co-variance matrix value(-10^12) and no optimization happens. The structure of the graph is preserved along the process and verified(dot file). The node/factor removal starts with the oldest node with id 0 and is incrementally deleted and works as expected and is verified by printing it to dot file. Is there any work around to this problem? or is there any other standard procedure to address this issue?

ProfFan, 2020-07-11:
Is this still an issue?

varunagrawal, 2023-09-25:
Closing due to lack of a response. @emil-jose please re-open if this is still an issue.

closed 2023-09-25 by varunagrawal