borglab/gtsam #10: iSAM2 and smartfactors

dellaert, 2019-05-19:
Re-opening [bitbucket issue 420](https://bitbucket.org/gtborg/gtsam/issues/420/problem-with-isam2-stereo-smart-factors-no)

jlblancoc, 2019-05-19:
@dellaert Brilliant! Next week, hopefully will come back to this important one. 
Cheers.

jlblancoc, 2019-05-25:
Work in progress [here](https://github.com/borglab/gtsam/compare/develop...jlblancoc:fix_10_StereoSmartFactors_iSAM2).

On [this](https://bitbucket.org/gtborg/gtsam/issues/420/problem-with-isam2-stereo-smart-factors-no):
>  One things is that we definitely want to be backwards compatible on update, including in wrappers.

Let me know if [this solution](https://github.com/borglab/gtsam/commit/8ec8def901d0097733e82ca5fbb1fccdd0d5e1f7) (100% backwards compatible) seems better.
Next, we can add to that structure the new parameters needed to solve this bug.

On this:
> “In theory, even smart-factors know which keys they are connected to, so I am not sure why simply giving the keys is not an option.”

It would be great to do it like that; last time I spent some time with this, though, it looked like it was not possible, but will give it another try just in case I missed something.

jlblancoc, 2019-05-27:
I'm super happy this is solved, yay!

@dellaert Please, take a look at the proposed solution in the PR #25. 

> “In theory, even smart-factors know which keys they are connected to, so I am not sure why simply giving the keys is not an option.”

Perhaps there's another way, but I though that for iSAM2::getAffectedFactors() to work properly: 

https://github.com/borglab/gtsam/blob/dc774ed8e93ef00454b79904e0afccc0440e2eba/gtsam/nonlinear/ISAM2.cpp#L109-L113

we need `variableIndex_` to be up-to-date, and that implies requesting from the user this information, that's the reason of the new parameter to `update()`.

dellaert, 2019-05-28:
Excellent, I will review...

PR #25 (2019-05-29): Fix iSAM2 support for smart factors
A new optional parameter has been added to iSAM2::update(), only needed when using smart factors, or other factors that depend on a dynamic number of Keys.

A new unit test has been added: it runs a toy Stereo SLAM problem with both, batch LM, and iSAM2. It has a "verbosity" macro that can be commented out to enable printing all the details. Originally, the test failed with an error of "ill posed problem", it works now with iSAM2 giving exactly the same result than LM.

Closes #10
closed 2019-05-29 by dellaert