borglab/gtsam #11: Boost warning

dellaert, 2019-05-19:
before tagging GTSAM 4, I want to get rid of boost rvalue warning. I think the issue is that we are sometimes allowing boost::optional of references, which BTW caused some segfaults for me in the past: to fix I had to allocate the optional thing on the stack first.

dellaert, 2019-05-30:
@chrisbeall do you remember the *reason* we allowed this unsafe behavior? We have optional Matrix& everywhere, but that's non-const. Hoping it's only an issue for const & as that is easier to fix (less of them).

dellaert, 2019-05-30:
Update: if I remove 
```
if(NOT (${Boost_VERSION} LESS 105600))
	message("Ignoring Boost restriction on optional lvalue assignment from rvalues")
	list(APPEND GTSAM_COMPILE_DEFINITIONS_PUBLIC BOOST_OPTIONAL_ALLOW_BINDING_TO_RVALUES BOOST_OPTIONAL_CONFIG_ALLOW_BINDING_TO_RVALUES)
endif()
```
from cmake files, all tests still succeed, which is surprising. Maybe not fully understanding....

chrisbeall, 2019-06-02:
We already had the troublesome usage of `boost::optional` in the code, and we did not run into trouble until I upgraded to the latest version of Boost which introduced the warnings and `BOOST_OPTIONAL_ALLOW_BINDING_TO_RVALUES`. Issue: [Can't build GTSAM with Boost 1.56 with C++11](https://bitbucket.org/gtborg/gtsam/issues/116/cant-build-gtsam-with-boost-156-with-c-11). 

It looks like the code here hasn't really changed since then: https://github.com/borglab/gtsam/blob/823c8bd5e848892a2cc6a7186243aee7f556dbe4/gtsam/inference/EliminateableFactorGraph-inst.h#L73-L78
Why does it work without `BOOST_OPTIONAL_ALLOW_BINDING_TO_RVALUES` now?
- I don't recall whether we saw this issue on every platform.
- Better compiler? (docs mention wrong behavior with reference to integral types, so not relevant here)
- Boost changes?

Relevant documentation: 
- https://www.boost.org/doc/libs/1_70_0/libs/optional/doc/html/boost_optional/dependencies_and_portability/optional_reference_binding.html
- https://www.boost.org/doc/libs/1_70_0/libs/optional/doc/html/boost_optional/optional_references.html

LiShuaixin, 2019-06-13:
@dellaert Hi, when I built the project it failed and showed redeclaration error as follow:

```
In file included from /usr/local/include/gtsam/base/timing.h:118:0,
                 from /usr/local/include/gtsam/inference/VariableIndex-inl.h:21,
                 from /usr/local/include/gtsam/inference/VariableIndex.h:195,
                 from /usr/local/include/gtsam/inference/Ordering.h:24,
                 from /usr/local/include/gtsam/inference/EliminateableFactorGraph.h:26,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:25,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:24,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizer.h:22,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:23,
                 from /home/lee/experiment/6-phd_program/include/factor_graph_optimization.h:20,
                 from /home/lee/experiment/6-phd_program/include/system.h:40,
                 from /home/lee/experiment/6-phd_program/src/system.cpp:1:
/usr/include/boost/timer/timer.hpp:32:1: error: ‘namespace boost::timer { }’ redeclared as different kind of symbol
 {
 ^
In file included from /usr/include/boost/progress.hpp:22:0,
                 from /home/lee/experiment/6-phd_program/include/surfel_map/synchronized_circular_buffer.h:11,
                 from /home/lee/experiment/6-phd_program/include/local_map.h:24,
                 from /home/lee/experiment/6-phd_program/include/mapping.h:33,
                 from /home/lee/experiment/6-phd_program/include/map_drawer.h:18,
                 from /home/lee/experiment/6-phd_program/include/system.h:34,
                 from /home/lee/experiment/6-phd_program/src/system.cpp:1:
/usr/include/boost/timer.hpp:44:7: note: previous declaration ‘class boost::timer’
 class timer
       ^
In file included from /usr/local/include/gtsam/base/timing.h:118:0,
                 from /usr/local/include/gtsam/inference/VariableIndex-inl.h:21,
                 from /usr/local/include/gtsam/inference/VariableIndex.h:195,
                 from /usr/local/include/gtsam/inference/Ordering.h:24,
                 from /usr/local/include/gtsam/inference/EliminateableFactorGraph.h:26,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:25,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:24,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizer.h:22,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:23,
                 from /home/lee/experiment/6-phd_program/include/factor_graph_optimization.h:20,
                 from /home/lee/experiment/6-phd_program/include/system.h:40,
                 from /home/lee/experiment/6-phd_program/src/system.cpp:1:
/usr/include/boost/timer/timer.hpp: In member function ‘std::__cxx11::string boost::timer::cpu_timer::format(short int, const string&) const’:
/usr/include/boost/timer/timer.hpp:68:34: error: ‘format’ is not a member of ‘boost::timer’
                         { return ::boost::timer::format(elapsed(), places, format); }
                                  ^
/usr/include/boost/timer/timer.hpp: In member function ‘std::__cxx11::string boost::timer::cpu_timer::format(short int) const’:
/usr/include/boost/timer/timer.hpp:70:34: error: ‘format’ is not a member of ‘boost::timer’
                         { return ::boost::timer::format(elapsed(), places); }
                                  ^
In file included from /usr/local/include/gtsam/inference/VariableIndex-inl.h:21:0,
                 from /usr/local/include/gtsam/inference/VariableIndex.h:195,
                 from /usr/local/include/gtsam/inference/Ordering.h:24,
                 from /usr/local/include/gtsam/inference/EliminateableFactorGraph.h:26,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:25,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:24,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizer.h:22,
                 from /usr/local/include/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:23,
                 from /home/lee/experiment/6-phd_program/include/factor_graph_optimization.h:20,
                 from /home/lee/experiment/6-phd_program/include/system.h:40,
                 from /home/lee/experiment/6-phd_program/src/system.cpp:1:
/usr/local/include/gtsam/base/timing.h: At global scope:
/usr/local/include/gtsam/base/timing.h:166:21: error: ‘cpu_timer’ in ‘class boost::timer’ does not name a type
       boost::timer::cpu_timer timer_;
                     ^
In file included from /usr/local/include/gtsam/base/timing.h:118:0,
                 from /usr/local/include/gtsam/inference/VariableIndex-inl.h:21,
                 from /usr/local/include/gtsam/inference/VariableIndex.h:195,
                 from /usr/local/include/gtsam/inference/Ordering.h:24,
                 from /usr/local/include/gtsam/inference/EliminateableFactorGraph.h:26,
                 from /usr/local/include/gtsam/linear/GaussianFactorGraph.h:25,
                 from /usr/local/include/gtsam/nonlinear/NonlinearOptimizerParams.h:24,
[... 34 lines cut ...]
                         { return ::boost::timer::format(elapsed(), places, format); }
                                  ^
[... cut ...]
                 from /home/lee/experiment/6-phd_program/include/mapping.h:22,
                 from /home/lee/experiment/6-phd_program/src/mapping.cpp:1:
/usr/local/include/gtsam/base/timing.h: At global scope:
/usr/local/include/gtsam/base/timing.h:166:21: error: ‘cpu_timer’ in ‘class boost::timer’ does not name a type
       boost::timer::cpu_timer timer_;
                     ^

```
The building environment is Ubuntu 16.04 with boost 1.5.8. I'm not familiar with boost, so could you please take a look on this issue.

ProfFan, 2019-09-23:
Getting this into 4.1.

ProfFan, 2019-09-23:
#121

ProfFan, 2019-12-20:
I think as #156 is merged this can be closed?

dellaert, 2019-12-20:
I think we still need to tell cmake about having boost disallow it, I did not see that in the PR. Would be nice as that is frequently the only warning.

ProfFan, 2019-12-20:
Great, I'll make a PR to remove the flag.

ProfFan, 2019-12-31:
Closed by #201.

closed 2019-12-31 by ProfFan