borglab/gtsam #8: Make check fails because of GeographicLib

dellaert, 2019-05-19:
make check seems to fail on Mac, with default options, because of a link error. I think this might have been the recent PR? Maybe it's my version? But cmake did not complain :-)

```
[ 74%] Building CXX object gtsam/navigation/tests/CMakeFiles/testGeographicLib.dir/testGeographicLib.cpp.o
[ 74%] Linking CXX executable testGeographicLib
Undefined symbols for architecture x86_64:
  "GeographicLib::Geocentric::WGS84()", referenced from:
      __GLOBAL__sub_I_testGeographicLib.cpp in testGeographicLib.cpp.o
  "GeographicLib::UTMUPS::EncodeZone(int, bool, bool)", referenced from:
      GeographicLibUTMTest::run(TestResult&) in testGeographicLib.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [gtsam/navigation/tests/testGeographicLib] Error 1
make[2]: *** [gtsam/navigation/tests/CMakeFiles/testGeographicLib.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....

```

chrisbeall, 2019-05-19:
Can you confirm that you have previously compiled/installed GeographicLib via gtsam, and that it's architecture is i386?

Unfortunately there are several factors here:
- GeographicLib used to target i386 on Mac. Latest version defaults to x86_64.
- Apple has deprecated building for i386. Older versions of Xcode happily built for i386, but the most recent version does not. (**Latest Xcode will *not* build previous versions of GeographicLib**)
- The default option of our `GTSAM_INSTALL_GEOGRAPHICLIB` is set to `ON` if library was *not* found on system, and `OFF` if library was found ([here](https://github.com/borglab/gtsam/blob/20f29fba0cdbd154a146b58786591be9896af5b2/gtsam/3rdparty/CMakeLists.txt#L51)). Not sure this is working, I always seem to end up with default `OFF`. 

I have macOS 10.14.5, and Xcode with `Apple LLVM version 10.0.1 (clang-1001.0.46.4)`, and I had a previously installed (old, i386) GeographicLib. I ran into problems linking against this old version, so I removed it, and then found that Xcode would not rebuild it. 

Best way is forward:
- Can you delete old installed version, and try again?
- We can add a CMake check for recent version and warnings/documentation.

dellaert, 2019-05-19:
OK, so, I'm assuming that this indeed because of an old install. However, now in Montreal with an old computer. I will investigate later tonight whether a fresh install solves it, but I think cmake should check for the right version, if possible.

dellaert, 2019-06-15:
This was probably a local install issue.

closed 2019-06-15 by dellaert