borglab/gtsam #19: build error

amiltonwong, 2019-05-23:
Hi, all,

I got the following build error:
```
[ 35%] Building CXX object gtsam/CMakeFiles/gtsam.dir/geometry/StereoCamera.cpp.o
[ 35%] Building CXX object gtsam/CMakeFiles/gtsam.dir/geometry/StereoPoint2.cpp.o
[ 38%] Building CXX object gtsam/CMakeFiles/gtsam.dir/geometry/Unit3.cpp.o
[ 38%] Building CXX object gtsam/CMakeFiles/gtsam.dir/geometry/triangulation.cpp.o
In file included from /data/code9/gtsam/gtsam/nonlinear/LevenbergMarquardtOptimizer.h:25:0,
                 from /data/code9/gtsam/gtsam/geometry/triangulation.cpp:22:
/data/code9/gtsam/gtsam/linear/VectorValues.h: In member function ‘std::pair<tbb::interface5::internal::solist_iterator<tbb::interface5::internal::split_ordered_list<std::pair<const long unsigned int, Eigen::Matrix<double, -1, 1> >, tbb::tbb_allocator<std::pair<const long unsigned int, Eigen::Matrix<double, -1, 1> > > >, std::pair<const long unsigned int, Eigen::Matrix<double, -1, 1> > >, bool> gtsam::VectorValues::tryInsert(gtsam::Key, const Vector&)’:
/data/code9/gtsam/gtsam/linear/VectorValues.h:200:22: error: ‘gtsam::VectorValues::Values {aka class gtsam::ConcurrentMap<long unsigned int, Eigen::Matrix<double, -1, 1> >}’ has no member named ‘emplace’
       return values_.emplace(j, value); 
                      ^
make[2]: *** [gtsam/CMakeFiles/gtsam.dir/geometry/triangulation.cpp.o] Error 1
make[1]: *** [gtsam/CMakeFiles/gtsam.dir/all] Error 2
make: *** [all] Error 2
```
Here is the [cmake log](https://gist.githubusercontent.com/amiltonwong/2a06c63451f2c4f2c3e4a56b1c44a40e/raw/408593d29acb5c9e88aea7af1d9c628429a16904/cmake_log).

Any hints to solve this problem?

THX!

chrisbeall, 2019-05-24:
Please try with TBB disabled? The flag is `GTSAM_WITH_TBB`. It looks like your TBB is a bit old (4.2).
My Ubuntu 16.04 machine has TBB 4.4, and that definitely works.

amiltonwong, 2019-05-24:
@chrisbeall , Thanks a lot. After disable it, building process succeeds.

chrisbeall, 2019-05-25:
Great! What version of Linux are you on? I suspect it's pretty old because TBB 4.2 was [released September 2013](https://software.intel.com/en-us/forums/intel-threading-building-blocks/topic/472327).
It looks like TBB became fully C++11 move-aware as of version [4.3 Update 6](https://github.com/intel/tbb/blob/tbb_2019/CHANGES#L1050), so we probably ought to make that version or 4.4 ([released August 2015](https://www.threadingbuildingblocks.org/news/intel-tbb-4-4-released)) the lowest version which gtsam supports.

amiltonwong, 2019-05-26:
@chrisbeall , my system is a bit old, it's Ubuntu 14.04.

dellaert, 2019-05-26:
Fixed by PR #23

closed 2019-05-26 by dellaert