borglab/gtsam #20: Smartfactor body_P_sensor

dellaert, 2019-05-23:
Not respected: fix it.

mikesheffler, 2019-06-12:
What's the deal with this? I see `body_P_sensor_` mentioned in the `unwhitenedError` method in `SmartFactorBase.h`, but I think that's it. Where is it missed?

dellaert, 2019-06-12:
Needs a unit test whether if we pass `body_P_sensor`, it is actually used in projection factors and the like. It seems it is not, so I expect unit test to fail, and then we have to fix that :-)

mikesheffler, 2019-06-20:
I swear I'm not trying to be dense, but is this not covered by

`TEST( SmartProjectionPoseFactor, smartFactorWithSensorBodyTransform )`

in `testSmartProjectionPoseFactor.cpp`?

I'm suddenly interested because I'm using `SmartProjectionPoseFactor<Cal3_S2>` with a `sensor_to_body` transform, and I'm having a hell of a time.

I'm sure I'm just putting some garbage data in somewhere, but I figure it's worth asking ... :grin:

mikesheffler, 2019-06-21:
Ah. Did you have [this](https://bitbucket.org/gtborg/gtsam/issues/411/smartposeprojfactor-error-with-sensor) (issue number 41 from the bitbucket site) in mind? What kind of test did you need?

dellaert, 2019-06-21:
Just a test that checks that the projections are being done correctly if the camera is offset. Including derivatives

ProfFan, 2020-07-11:
Closed by #379, please reopen if not :)

closed 2020-07-11 by ProfFan