borglab/gtsam #22: Error when using find_package(GTSAM) inside CMakeLists.txt

itzsid, 2019-05-26:
I cloned the latest develop branch of GTSAM on Ubuntu 16.04. 
CMakeLists.txt looks like this: 
```
cmake_minimum_required(VERSION 3.0.0)
project(gtsam_test)
find_package(GTSAM REQUIRED)
```

Error is in `find_dependency()` inside GTSAMConfig.cmake. The error is as follows:
```
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /usr/local/lib/cmake/GTSAM/GTSAMConfig.cmake:17 (find_dependency)
  CMakeLists.txt:3 (find_package)
```

Seems to be related to this [issue](https://github.com/pabloariasal/modern-cmake-sample/issues/5).

dellaert, 2019-05-26:
Interesting. Should we bump the cmake version requirement?

dellaert, 2019-05-26:
I’m not a cmake expert, maybe @jlblancoc wants to make a recommendation?

chrisbeall, 2019-05-26:
Does it work if you replace `find_dependency` with `find_package`? That's one of the suggestions in the issues you linked.

CMake 3.5.1 is the default on Ubuntu 16.04, which we should still support. I vote against bumping the requirement to CMake 3.8.

jlblancoc, 2019-05-26:
Sure. We should find an alternative to keep support for u16.04...

@itzsid Please, give a try to [this patch](https://github.com/borglab/gtsam/pull/24/files), and confirm if it solves the problem, before merging. 
Thanks

itzsid, 2019-05-27:
@jlblancoc The [patch](https://github.com/borglab/gtsam/pull/24/files) works for me. Thanks!

PR #24 (2019-05-27): Fix GTSAMconfig.cmake for cmake <3.8
Closes #22
commit a4b5f4cf (2019-05-27): Fix GTSAMconfig.cmake for cmake <3.8

Closes #22
closed 2019-05-27 by chrisbeall