borglab/gtsam #30: Documentation of factors

yetongatech, 2019-05-30:
May I know if there is a documentation of all the factors? @dellaert

dellaert, 2019-05-30:
I agree this is lacking. I have been wanting to improve doxygen main page.

isaiahoh, 2024-09-28:
Hi @dellaert,
I've started working on improved documentation for factors on my fork of GTSAM. I'm using Sphinx connected to Doxygen in a Read the Docs format. This approach allows for better organization of related terms like Factors without affecting the current documentation structure.
The Read the Docs format with Sphinx offers several advantages:
1. Easy navigation with a sidebar for quick access to different sections
2. Search functionality for finding specific topics

I've attached a screenshot below for a first look at the new documentation structure in the Read the Docs format. I'd appreciate your feedback on this approach and the initial layout. Let me know if this direction could be valuable for the project.
[image]