borglab/gtsam #34: Rethink gradient values for Constrained Factors

varunagrawal, 2019-05-31:
The behavior of gradients, specifically `gradientAtZero` methods for constrained factors/noise models needs to be re-evaluated. Concretely, gradients for constrained factors should be `undefined` since they are point functions with 0 covariance.

This behavior is not respected in the BayesTree data structure since while the constrained factors set the gradient to zero, the BayesTree aggregates the gradients from other factors on a variable without considering the constraints.

dellaert, 2019-05-31:
Note: this is a bug as it makes iSAM2 with Dogleg do the wrong thing for constrained factors.

ProfFan, 2019-09-23:
Should this be in 4.1? @dellaert

dellaert, 2019-09-23:
Let's discuss w @varunagrawal

ProfFan, 2019-09-26:
In my recent bughunting I got really confused about this piece of code in `NoiseModel.cpp`:

```
namespace internal {
// switch precisions and invsigmas to finite value
// TODO: why?? And, why not just ask s==0.0 below ?
static void fix(const Vector& sigmas, Vector& precisions, Vector& invsigmas) {
  for (Vector::Index i = 0; i < sigmas.size(); ++i)
    if (!std::isfinite(1. / sigmas[i])) {
      precisions[i] = 0.0;
      invsigmas[i] = 0.0;
    }
}
}
```

If the sigma is 0, invsigma should be `Inf` instead of 0.

Also note that division by zero is actually UB in C++.

dellaert, 2019-09-26:
Yeah... Where is that called from?

ProfFan, 2019-09-26:
@dellaert 
```c++
Constrained::Constrained(const Vector& sigmas)
  : Diagonal(sigmas), mu_(Vector::Constant(sigmas.size(), 1000.0)) {
  internal::fix(sigmas, precisions_, invsigmas_);
}
```

ProfFan, 2019-12-23:
#121

ProfFan, 2020-02-19:
@dellaert Ping