borglab/gtsam #39: Travis fails but shows success

dellaert, 2019-06-01:
As an example, [this run]( https://travis-ci.com/borglab/gtsam/jobs/204686142) did not run the tests but failed (why?, was just a warning), but still shows up as a success.

dellaert, 2019-06-01:
Travis is awesome ! We'll get it right :-)

jlblancoc, 2019-06-01:
Hmm... I think the point is that Travis build #5, [here](https://travis-ci.com/borglab/gtsam/builds/113966855), spans into the matrix of jobs we defined and only one of jobs failed, [this one, 5.2](https://travis-ci.com/borglab/gtsam/jobs/204686143)... in this case due to a timeout (50 minutes) :man_facepalming: 

Any of the admins, could try hitting the "retry" button of [job 5.2](https://travis-ci.com/borglab/gtsam/jobs/204686143), since sometimes build times differ depending on the load of the server. 

Anyhow, it seems we'll have to use Travis CI's "stages" to overcome the time limit (sigh)

dellaert, 2019-06-03:
Fixed issues in pr  #45

closed 2019-06-03 by dellaert