borglab/gtsam #41: testDoglegOptimizer fails with TBB

chrisbeall, 2019-06-02:
Fails on Ubuntu 16.04 with TBB 4.4.

```155/236 Test #155: testDoglegOptimizer ....................***Failed    0.01 sec
TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
/home/cbeall/git/gtsam/tests/testDoglegOptimizer.cpp:184: Failure: "Wrong exception: optimizer2.optimize(), expected: std::invalid_argument" 
There were 1 failures
```

chrisbeall, 2019-06-02:
Also fails on Mac with TBB 2019.0.

dellaert, 2019-06-02:
I think this is because TBB changes the exception somehow, see [this post](https://software.intel.com/en-us/blogs/2014/08/28/exact-exception-propagation-in-the-intel-threading-building-blocks-library-and-c11?_ga=2.99521434.567067728.1559484676-27613893.1559484676).

LiShuaixin, 2019-06-11:
Hi, have you solved this issue? I got the same problem... If you just use `make` instead of `make check`, the building could be done. But I'm not sure whether this way of building  may occur other issues, since it throw off segment fault in my project. [(this post)](https://github.com/borglab/gtsam/issues/50)

dellaert, 2019-06-11:
Fixed in PR #59

closed 2019-06-11 by dellaert