borglab/gtsam #43: Build issue with GTSAM 4.0

thinhhoang95, 2019-06-02:
I've just built GTSAM 4.0 with VC++ 14.1 on VS 2017. During full project build, I get the error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in libboost. It turns out that I need to remove libboost release dlls (those do not contain "gd" in their filename) in Linker > Input > Additional dependencies. I wonder if this is a bug. Also, 8 projects failed to build with the following error, I cannot seem to trace the reason why. 
`Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2005	"public: static class Eigen::CwiseNullaryOp<struct Eigen::internal::scalar_constant_op<double>,class Eigen::Matrix<double,2,1,0,2,1> > const __cdecl Eigen::DenseBase<class Eigen::Matrix<double,2,1,0,2,1> >::Zero(void)" (?Zero@?$DenseBase@V?$Matrix@N$01$00$0A@$01$00@Eigen@@@Eigen@@SA?BV?$CwiseNullaryOp@U?$scalar_constant_op@N@internal@Eigen@@V?$Matrix@N$01$00$0A@$01$00@3@@2@XZ) already defined in Event.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: class Eigen::Matrix<double,3,1,0,3,1> & __thiscall Eigen::EigenBase<class Eigen::Matrix<double,3,1,0,3,1> >::derived(void)" (?derived@?$EigenBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QAEAAV?$Matrix@N$02$00$0A@$02$00@2@XZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: class Eigen::Matrix<double,3,1,0,3,1> const & __thiscall Eigen::EigenBase<class Eigen::Matrix<double,3,1,0,3,1> >::derived(void)const " (?derived@?$EigenBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEABV?$Matrix@N$02$00$0A@$02$00@2@XZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: class Eigen::Matrix<double,3,1,0,3,1> & __thiscall Eigen::EigenBase<class Eigen::Matrix<double,3,1,0,3,1> >::const_cast_derived(void)const " (?const_cast_derived@?$EigenBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEAAV?$Matrix@N$02$00$0A@$02$00@2@XZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::DenseCoeffsBase<class Eigen::Matrix<double,3,1,0,3,1>,3>::outerStride(void)const " (?outerStride@?$DenseCoeffsBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@$02@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: static class Eigen::CwiseNullaryOp<struct Eigen::internal::scalar_constant_op<double>,class Eigen::Matrix<double,3,1,0,3,1> > const __cdecl Eigen::DenseBase<class Eigen::Matrix<double,3,1,0,3,1> >::Zero(void)" (?Zero@?$DenseBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@SA?BV?$CwiseNullaryOp@U?$scalar_constant_op@N@internal@Eigen@@V?$Matrix@N$02$00$0A@$02$00@3@@2@XZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::PlainObjectBase<class Eigen::Matrix<double,3,1,0,3,1> >::rows(void)const " (?rows@?$PlainObjectBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::PlainObjectBase<class Eigen::Matrix<double,3,1,0,3,1> >::cols(void)const " (?cols@?$PlainObjectBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: double const * __thiscall Eigen::PlainObjectBase<class Eigen::Matrix<double,3,1,0,3,1> >::data(void)const " (?data@?$PlainObjectBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEPBNXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: void __thiscall Eigen::PlainObjectBase<class Eigen::Matrix<double,3,1,0,3,1> >::resize(int,int)" (?resize@?$PlainObjectBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QAEXHH@Z) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: static void __cdecl Eigen::PlainObjectBase<class Eigen::Matrix<double,3,1,0,3,1> >::_check_template_params(void)" (?_check_template_params@?$PlainObjectBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@SAXXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::Matrix<double,3,1,0,3,1>::outerStride(void)const " (?outerStride@?$Matrix@N$02$00$0A@$02$00@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"protected: __thiscall Eigen::MatrixBase<class Eigen::Matrix<double,3,1,0,3,1> >::MatrixBase<class Eigen::Matrix<double,3,1,0,3,1> >(void)" (??0?$MatrixBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@IAE@XZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::EigenBase<class Eigen::Matrix<double,3,1,0,3,1> >::rows(void)const " (?rows@?$EigenBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::EigenBase<class Eigen::Matrix<double,3,1,0,3,1> >::cols(void)const " (?cols@?$EigenBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: double & __thiscall Eigen::PlainObjectBase<class Eigen::Matrix<double,3,1,0,3,1> >::coeffRef(int)" (?coeffRef@?$PlainObjectBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QAEAANH@Z) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2005	"public: int __thiscall Eigen::EigenBase<class Eigen::Matrix<double,3,1,0,3,1> >::size(void)const " (?size@?$EigenBase@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@Eigen@@QBEHXZ) already defined in SimPolygon2D.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\gtsamDebug.lib(gtsamDebug.dll)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Dummy.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\libboost_thread-vc141-mt-x32-1_70.lib(tss_pe.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in Dummy.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\libboost_thread-vc141-mt-x32-1_70.lib(tss_pe.obj)	1	
Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Dummy.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\libboost_thread-vc141-mt-x32-1_70.lib(thread.obj)	1	
Error	LNK2038	mismatch detected for 'RuntimeLibrary': value 'MD_DynamicRelease' doesn't match value 'MDd_DynamicDebug' in Dummy.obj	gtsam_unstable	D:\gtsam-4.0.0\build\gtsam_unstable\libboost_thread-vc141-mt-x32-1_70.lib(thread.obj)	1	
Error	LNK1169	one or more multiply defined symbols found	gtsam_unstable	D:\gtsam-4.0.0\build\bin\Debug\gtsam_unstableDebug.dll	1	
Error	LNK2001	unresolved external symbol "public: virtual void __thiscall gtsam::PreintegratedCombinedMeasurements::resetIntegration(void)" (?resetIntegration@PreintegratedCombinedMeasurements@gtsam@@UAEXXZ)	ISAM2_SmartFactorStereo_IMU	D:\gtsam-4.0.0\build\examples\ISAM2_SmartFactorStereo_IMU.obj	1	
Error	LNK2001	unresolved external symbol "public: virtual void __thiscall gtsam::PreintegratedCombinedMeasurements::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)const " (?print@PreintegratedCombinedMeasurements@gtsam@@UBEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)	ISAM2_SmartFactorStereo_IMU	D:\gtsam-4.0.0\build\examples\ISAM2_SmartFactorStereo_IMU.obj	1	
[... cut ...]
Error	C2027	use of undefined type 'gtsam::traits<T>'	SmartRangeExample_plaza2	D:\gtsam-4.0.0\gtsam\base\GenericValue.h	152	
Error	C3861	'GetDimension': identifier not found	SmartRangeExample_plaza2	D:\gtsam-4.0.0\gtsam\base\GenericValue.h	152	
Error	C2678	binary '=': no operator found which takes a left-hand operand of type 'T' (or there is no acceptable conversion)	SmartStereoProjectionFactorExample	D:\gtsam-4.0.0\gtsam\base\GenericValue.h	171	
`
May I have some suggestions on how to resolve this? Many thanks!

jlblancoc, 2019-06-02:
Where did you get Boost from? Custom build? Binary install?

thinhhoang95, 2019-06-02:
I got Boost from binary for Windows

jlblancoc, 2019-06-08:
Please, give it a try to the code version in this PR: https://github.com/borglab/gtsam/pull/27

It should have fixed those errors.

dellaert, 2019-06-11:
@thinhhoang95 please follow up or close the issue if now resolved?

thinhhoang95, 2019-06-12:
I'm trying the solution and will get back in a few hours

thinhhoang95, 2019-06-23:
Sorry for the late reply. I confirm that it was a bug in VC++ 2017.

closed 2019-06-23 by thinhhoang95