borglab/gtsam #50: RUN issue

LiShuaixin, 2019-06-10:
Hello, thanks for your patience. I cloned the latest develop branch of GTSAM on Ubuntu 16.04 with Boost 1.58 and tried importing it in my project. 

Part of CMakeLists.txt looks like this:
`find_package(Boost COMPONENTS date_time serialization program_options thread system signals regex filesystem REQUIRED )

find_package( Eigen3 REQUIRED )

find_package( GTSAM REQUIRED )

include_directories( 
${PROJECT_SOURCE_DIR}
${PROJECT_SOURCE_DIR}/include
${Boost_INCLUDE_DIRS}
${EIGEN3_INCLUDE_DIR} 
${GTSAM_INCLUDE_DIR}
${TBB_INCLUDE_DIRS}
)

add_library(${PROJECT_NAME} SHARED
src/factor_graph.cpp)

target_link_libraries(${PROJECT_NAME}
${EIGEN3_LIBS}
${GTSAM_LIBRARY_DIRS}
/usr/lib/x86_64-linux-gnu/libtbb.so
)

target_link_libraries(${PROJECT_NAME}
${Boost_LIBRARIES} -lgtsam -ltbb)`

factor_graph.cpp is a source code file where I put all factors I need.

It worked smoothly and I got a executable. But when I run it, the system broke down, and throw off segment fault directly before step into the main function in  executable.cpp file. Then I commented the line `src/factor_graph.cpp` in CMakeLists and rebuilt it. The executable could run again.

did I import gtsam right? please help. thanks in advance!

jlblancoc, 2019-06-10:
Could you give it a try to this branch [1]?
That's exactly the kind of problems that those changes are trying to
avoid... Please, report back whether it worked or not.

JL

[1] https://github.com/jlblancoc/gtsam/tree/flags_via_modern_cmake

LiShuaixin, 2019-06-10:
Hi @jlblancoc , thanks for your reply! I will give it a try and report back. 

Actually, I can smoothly run [lego-loam](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM) which uses gtsam to achieve global optimization. I wonder why it fail in my project while success in lego-loam on the same PC

LiShuaixin, 2019-06-10:
Hi, @jlblancoc. I tried rebuilding the branch version, but failed when `make check`. 

Errors output as follow:
```
99% tests passed, 1 tests failed out of 236

Total Test time (real) =   2.44 sec

The following tests FAILED:
	166 - testDoglegOptimizer (Failed)
Errors while running CTest
CMakeFiles/check.dir/build.make:57: recipe for target 'CMakeFiles/check' failed
make[3]: *** [CMakeFiles/check] Error 8
CMakeFiles/Makefile2:651: recipe for target 'CMakeFiles/check.dir/all' failed
make[2]: *** [CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:658: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:221: recipe for target 'check' failed
make: *** [check] Error 2
```

Then I rebuilt it by 'make -j4' directly, and it worked.

But when I imported it in my project using the same cmakelist file, errors occurred since GTSAM was built against a different version of Eigen:
```
In file included from /usr/local/include/gtsam/base/Matrix.h:26:0,
                 from /usr/local/include/gtsam/base/Manifold.h:22,
                 from /usr/local/include/gtsam/base/Lie.h:25,
                 from /usr/local/include/gtsam/base/VectorSpace.h:11,
                 from /usr/local/include/gtsam/geometry/Point2.h:20,
                 from /usr/local/include/gtsam/geometry/Unit3.h:23,
                 from /usr/local/include/gtsam/geometry/Rot3.h:25,
                 from /home/lee/loam_ws/src/Inertial-LOAM/Inertial-LOAM/src/mapOptmization.cpp:36:
/usr/local/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
```

jlblancoc, 2019-06-10:
Good, it seems that the crash was caused by mixing Eigen versions (probably)... it's better to have a build error than a runtime crash, IMO. 

Try rebuilding GTSAM changing its CMake variables such as it uses the system version of Eigen, assuming that your own project does so as well, make install again, and check if it works.

LiShuaixin, 2019-06-11:
Could you give me a brief instruction on how to change its Eigen verision when rebuilding GTSAM? I'm not a expert on CMake.

jlblancoc, 2019-06-11:
run `cmake-gui .` from your build directory, check "GTSAM_USE_SYSTEM_EIGEN", configure, generate, rebuild from the console as usual.

LiShuaixin, 2019-06-11:
I tried both verision of Eigen and got the same error when run `make check` to build GTSAM. 
```
TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
/home/lee/gtsam-develop/tests/testDoglegOptimizer.cpp:184: Failure: "Wrong exception: optimizer2.optimize(), expected: std::invalid_argument" 
There were 1 failures
```
It should be something related to TBB, which is the same issue in [#41](https://github.com/borglab/gtsam/issues/41)

LiShuaixin, 2019-06-11:
Hi @jlblancoc, @dellaert, I tried both 2 latest branches of GTSAM with system version of Eigen 3.3.4 or not(Eigen3.3.7), but got the same errors.

1. When built GTSAM, there was a TBB warning which seems that GTSAM is built without support for exact exception propagation. 
```
TBB Warning: Exact exception propagation is requested by application but the linked library is built without support for it
/home/lee/gtsam-flags_via_modern_cmake/tests/testDoglegOptimizer.cpp:184: Failure: "Wrong exception: optimizer2.optimize(), expected: std::invalid_argument" 
There were 1 failures
```

2. Ignoring this checking error and making install directly, the compatibility issue of Eigen version occurred when building my own project, which makes it failed importing the GTSAM lib.
```
In file included from /usr/local/include/gtsam/base/Matrix.h:26:0,
                 from /usr/local/include/gtsam/base/Manifold.h:22,
                 from /usr/local/include/gtsam/geometry/BearingRange.h:21,
                 from /usr/local/include/gtsam/geometry/Pose3.h:22,
                 from /usr/local/include/gtsam/navigation/NavState.h:21,
                 from /usr/local/include/gtsam/navigation/ManifoldPreintegration.h:24,
                 from /usr/local/include/gtsam/navigation/CombinedImuFactor.h:25,
                 from /home/lee/experiment/6-phd_program/include/factor_graph_optimization.h:14,
                 from /home/lee/experiment/6-phd_program/src/factor_graph_optimization.cpp:1:
/usr/local/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^
```

I also tried a previous version of GTSAM you shared before. Both of these 2 issues listed above were solved, but segment fault was thrown off as soon as system launching.

closed 2019-06-12 by LiShuaixin