borglab/gtsam #51: Feature Request: Make uninstall option

dwisth, 2019-06-10:
Hello,

Just a feature request. It would be great to have a `make uninstall` option. I uninstall gtsam (e.g. to upgrade to a new version) by using ```rm `cat install_manifest.txt` ``` but an uninstall option would be helpful.

Thanks!
David

jlblancoc, 2019-06-10:
Here you go: https://github.com/borglab/gtsam/pull/52
Please, give it a try and let us know if it works as expected (it should). 

Anyway: avoid doing "make install" as much as possible. CMake always allows you to use gtsam or any other project **without** installing it to any special path.

dwisth, 2019-06-10:
Thanks for the quick response!

I just tested it on my computer and it works perfectly.

jlblancoc, 2019-06-11:
Cool. Closing. 

(cc: @dellaert, @chrisbeall  )

PR #52 (2019-06-11): Provide target `make uninstall`
Adds a custom target to uninstall after doing "make install". 

Closes #51
closed 2019-06-11 by jlblancoc