borglab/gtsam #53: FTBFS when using TBB

jlblancoc, 2019-06-10:
It seems there are now build errors when TBB is enabled: 

```
gtsam/linear/VectorValues.h:200:22: error: no member named 'emplace' in 'gtsam::ConcurrentMap<unsigned long, Eigen::Matrix<double, -1, 1, 0, -1, 1> >'
      return values_.emplace(j, value); 
             ~~~~~~~ ^
```

due to the selection of `CONCURRENT_MAP_BASE`.

What's the preferred solution here? For now, disabling TBB is the only workaround.

dellaert, 2019-06-10:
Hmm. Fix might be as simple as `values_.insert(make_pair(j,value))` ?

dellaert, 2019-06-10:
Maybe guarded by TBB flag, as emplace is more efficient I think

chrisbeall, 2019-06-10:
I bumped up the minimum required version of TBB in https://github.com/borglab/gtsam/pull/23 because someone already reported this issue in https://github.com/borglab/gtsam/issues/19. What version of TBB do you have?

jlblancoc, 2019-06-10:
Hmm... ok, yes, I had the CMake warning but just ignored it.... my fault. 

Apparently, [this](https://packages.ubuntu.com/source/bionic/tbb) is my version but cannot find any "4.x" anything on that page (?). 
But this file `/usr/lib/x86_64-linux-gnu/libtbb.so.2` seems to mean Bionic comes with version 2? Or am I totally missing some point?

dellaert, 2019-06-10:
Hmmm, maybe if the fix above works there is no need to bump?

chrisbeall, 2019-06-10:
No, you don't have version 2. Ubuntu 16.04 has default version 4.4, and Ubuntu 18.04 has version 2017+. The suffix on libraries often has nothing to do with version numbers.

jlblancoc, 2019-06-11:
Hmm... should then I apply [this proposed patch](https://github.com/borglab/gtsam/issues/53#issuecomment-500473170) if using TBB? 
@chrisbeall Does this fail for you too when building with TBB?

chrisbeall, 2019-06-11:
Building with TBB works fine for me on Ubuntu 16.04 with TBB 4.4, and on Mac with TBB 2019-U5. I can't think of any reason why this would not work on Ubuntu 18.04 with TBB 2017-U7, but I've not yet tried.

[Here](https://github.com/intel/tbb/tags) you can see that TBB changed their versioning scheme. After 4.4.x they switched to using year + update #. 
The only issue I ran into is the `testDoglegOptimizer` test failure, which I reported in issue https://github.com/borglab/gtsam/issues/41

chrisbeall, 2019-06-11:
I just tried from scratch on Ubuntu 18.04 with TBB 2017-U7 on latest develop (4e19ba4488ea3cc93ee7d021fffde87efa1ac44f) , and everything compiled. The version of TBB you have clearly supports move-semantics, so there must be some misconfiguration or bug somewhere. Let's find the root cause, please?

jlblancoc, 2019-06-12:
Also updated to the latest develop commit and it builds ok.
Closing, sorry for the noise!

closed 2019-06-12 by jlblancoc