borglab/gtsam #57: Unit tests Segfault w clang on Linux

dellaert, 2019-06-11:
clang Linux in Release has segfaults in 4 unit tests. Seems serialization related. I am not running clang on Linux so would appreciate some help from someone who does or has a docker image handy.

jlblancoc, 2019-06-11:
This is what valgrind says: 

```
make testSerializationLinear.valgrind
[  4%] Built target CppUnitLite
[ 36%] Built target metis
[100%] Built target gtsam
[100%] Built target testSerializationLinear
Scanning dependencies of target testSerializationLinear.valgrind
==13410== Memcheck, a memory error detector
==13410== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==13410== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==13410== Command: /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear
==13410== 
==13410== Invalid read of size 8
==13410==    at 0x539BCC: void boost::archive::detail::save_pointer_type<boost::archive::text_oarchive>::polymorphic::save<gtsam::noiseModel::Diagonal>(boost::archive::text_oarchive&, gtsam::noiseModel::Diagonal&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x539B5F: boost::archive::detail::oserializer<boost::archive::text_oarchive, boost::shared_ptr<gtsam::noiseModel::Diagonal> >::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x549BA42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==13410==    by 0x547EF0: boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::JacobianFactor>::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x549BA42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==13410==    by 0x5505E8: boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::GaussianConditional>::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x549BA42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==13410==    by 0x55907A: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > gtsam::serialize<gtsam::GaussianConditional>(gtsam::GaussianConditional const&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x5241C7: bool gtsam::serializationTestHelpers::equalsObj<gtsam::GaussianConditional>(gtsam::GaussianConditional const&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x51D9CD: Serializationgaussian_conditionalTest::run(TestResult&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x566624: TestRegistry::run(TestResult&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x5221C7: main (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==13410== 
==13410== 
==13410== Process terminating with default action of signal 11 (SIGSEGV)
==13410==  Access not within mapped region at address 0x0
==13410==    at 0x539BCC: void boost::archive::detail::save_pointer_type<boost::archive::text_oarchive>::polymorphic::save<gtsam::noiseModel::Diagonal>(boost::archive::text_oarchive&, gtsam::noiseModel::Diagonal&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x539B5F: boost::archive::detail::oserializer<boost::archive::text_oarchive, boost::shared_ptr<gtsam::noiseModel::Diagonal> >::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x549BA42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==13410==    by 0x547EF0: boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::JacobianFactor>::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x549BA42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==13410==    by 0x5505E8: boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::GaussianConditional>::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x549BA42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==13410==    by 0x55907A: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > gtsam::serialize<gtsam::GaussianConditional>(gtsam::GaussianConditional const&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x5241C7: bool gtsam::serializationTestHelpers::equalsObj<gtsam::GaussianConditional>(gtsam::GaussianConditional const&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==    by 0x51D9CD: Serializationgaussian_conditionalTest::run(TestResult&) (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
[... cut ...] linear/tests/testSerializationLinear)
==13410==    by 0x5221C7: main (in /home/jlblanco/code/gtsam-jlblancoc/build-clang/gtsam/linear/tests/testSerializationLinear)
==13410==  If you believe this happened as a result of a stack
==13410==  overflow in your program's main thread (unlikely but
==13410==  possible), you can try to increase the size of the
==13410==  main thread stack using the --main-stacksize= flag.
==13410==  The main thread stack size used in this run was 8388608.
==13410== 

```

dellaert, 2019-06-12:
Don't see anything wrong with the code. Seems to bomb on serializing an optional. Might be a boost bug??

jlblancoc, 2019-06-16:
Some new tests, and a backtrace with better function names (-g): 

The new finding is: the error only happen with optimization levels -O1 or above.
- clang6, Debug build. Flags: -g;-fno-inline  ==> Runs OK.
- Flags: -g;-O0 ==> Ok.
- Flags: -g;-O1 ==> crashes.
- Release, -O3 ==> crashes

```
==31720== Invalid read of size 8
==31720==    at 0x75E680: boost::serialization::extended_type_info_typeid<gtsam::noiseModel::Diagonal>::get_derived_extended_type_info(gtsam::noiseModel::Diagonal const&) const (extended_type_info_typeid.hpp:107)
==31720==    by 0x75E578: void boost::archive::detail::save_pointer_type<boost::archive::text_oarchive>::polymorphic::save<gtsam::noiseModel::Diagonal>(boost::archive::text_oarchive&, gtsam::noiseModel::Diagonal&) (oserializer.hpp:392)
==31720==    by 0x75E409: boost::archive::text_oarchive& boost::archive::detail::interface_oarchive<boost::archive::text_oarchive>::operator<< <gtsam::noiseModel::Diagonal const*>(gtsam::noiseModel::Diagonal const* const&) (interface_oarchive.hpp:70)
==31720==    by 0x75E359: void boost::serialization::serialize_adl<boost::archive::text_oarchive, boost::serialization::nvp<gtsam::noiseModel::Diagonal const*> >(boost::archive::text_oarchive&, boost::serialization::nvp<gtsam::noiseModel::Diagonal const*>&, unsigned int) (serialization.hpp:126)
==31720==    by 0x75E269: boost::archive::text_oarchive& boost::archive::detail::interface_oarchive<boost::archive::text_oarchive>::operator<< <boost::serialization::nvp<gtsam::noiseModel::Diagonal const*> >(boost::serialization::nvp<gtsam::noiseModel::Diagonal const*> const&) (interface_oarchive.hpp:70)
==31720==    by 0x75E235: void boost::serialization::save<boost::archive::text_oarchive, gtsam::noiseModel::Diagonal>(boost::archive::text_oarchive&, boost::shared_ptr<gtsam::noiseModel::Diagonal> const&, unsigned int) (shared_ptr.hpp:106)
==31720==    by 0x75E1E9: boost::serialization::free_saver<boost::archive::text_oarchive, boost::shared_ptr<gtsam::noiseModel::Diagonal> >::invoke(boost::archive::text_oarchive&, boost::shared_ptr<gtsam::noiseModel::Diagonal> const&, unsigned int) (split_free.hpp:45)
==31720==    by 0x75E179: void boost::serialization::serialize_adl<boost::archive::text_oarchive, boost::shared_ptr<gtsam::noiseModel::Diagonal> >(boost::archive::text_oarchive&, boost::shared_ptr<gtsam::noiseModel::Diagonal>&, unsigned int) (serialization.hpp:126)
==31720==    by 0x75E0AC: boost::archive::detail::oserializer<boost::archive::text_oarchive, boost::shared_ptr<gtsam::noiseModel::Diagonal> >::save_object_data(boost::archive::detail::basic_oarchive&, void const*) const (oserializer.hpp:150)
==31720==    by 0x7233A42: ??? (in /usr/lib/x86_64-linux-gnu/libboost_serialization.so.1.65.1)
==31720==    by 0x75DE99: boost::archive::text_oarchive& boost::archive::detail::interface_oarchive<boost::archive::text_oarchive>::operator<< <boost::shared_ptr<gtsam::noiseModel::Diagonal> >(boost::shared_ptr<gtsam::noiseModel::Diagonal> const&) (interface_oarchive.hpp:70)
==31720==    by 0x75DDE9: void boost::serialization::serialize_adl<boost::archive::text_oarchive, boost::serialization::nvp<boost::shared_ptr<gtsam::noiseModel::Diagonal> > >(boost::archive::text_oarchive&, boost::serialization::nvp<boost::shared_ptr<gtsam::noiseModel::Diagonal> >&, unsigned int) (serialization.hpp:126)
==31720==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==31720== 
```

dellaert, 2019-06-16:
Hmmm, I hate saying this, but could be a Compiler bug - boost template magic tends to be intricate. We might have to special case this, though I don’t know how to deal with a nullptr in serialization.

ProfFan, 2019-09-23:
Will try to reproduce and fix. Also, should we add the latest compilers to Travis? Opt and Terra targets LLVM 8 already.

dellaert, 2019-09-23:
Nope. Different audience.

ProfFan, 2019-09-23:
Copy that :)

ProfFan, 2019-12-21:
I think this can be closed now.

jlblancoc, 2019-12-22:
In fact... I'm afraid it's NOT solved yet, the test is commented out in travis: 

https://github.com/borglab/gtsam/blob/087221ac955c9d66a15c07f383562dec64dc38ea/.travis.yml#L114-L119

ProfFan, 2019-12-22:
Errrgh... I'll make a PR for this...

varunagrawal, 2020-07-07:
@ProfFan I believe this has been fixed since the corresponding tests on Travis pass. Can you please confirm, and if yes, close this issue?

ProfFan, 2020-07-09:
Should have been fixed, we can reopen if this is not the case.

closed 2020-07-09 by ProfFan