borglab/gtsam #80: Improve plot3DTrajectory in Matlab

scharalambous3, 2019-06-29:
Hello,

As seen at https://bitbucket.org/gtborg/gtsam/issues/145/plot3dtrajectory-in-matlab, plot3DTrajectory in MATLAB is slow. For your reference, you can test it yourself by running Pose3SLAMExample_graph

I am thus trying to transform the use of for-loops to matrix manipulations.
For this to be achieved, I need to change the KeyVector.at(index) and Values.at functions(index) to work the same way on matrix input as well

The function calls are 
`gtsam_wrapper(1097, this, varargin{:})`
 and
`gtsam_wrapper(1037, this, varargin{:})` respectively. 
I am looking for some help on how to change these two functions (e.g. direct me to the respective functions in a .cpp file)

Thanks in advance,
Stefanos C.

dellaert, 2019-07-30:
Does this need a change to the wrapper?

scharalambous3, 2019-08-04:
As it can be seen in the attached profiling summary, I couldn't  make any significant improvements as I am restricted by the wrapper

Yes, specificially the following functions should be changed to accept matrix input as well:
Values.at, KeyVector.At as well as the children functions of gtsam.Pose3 (.x(), .y(), .z(), .rotation() etc)

 [image]

scharalambous3, 2019-08-05:
I will keep working on it, and proceed with changes in the wrapper if needed
Best,
Stefanos C.

dellaert, 2019-08-05:
I think this might be a case for a specific utility method. There is a utilities file that gets wrapped where a vector-based method could be easy to add.