borglab/gtsam #82: cmake error when building with MATLAB flag ON

izzys, 2019-07-10:
CMake Error at cmake/GtsamMatlabWrap.cmake:106 (message):
  You must call find_package(GTSAM) before using wrap
Call Stack (most recent call first):
  cmake/GtsamMatlabWrap.cmake:59 (wrap_library_internal)
  gtsam/CMakeLists.txt:214 (wrap_and_install_library)

varunagrawal, 2019-07-10:
This seems like GTSAM was either not built first or is installed somewhere which is not on you `PATH`.

Which OS are you on?

izzys, 2019-07-10:
Linux 16.04

I had a gtsam installation without MATLAB that went fine. Then I tried to build again with the MATLAB flag ON, and the build failed.

So I cleaned the previous installation, removed all files from usr/local and started all over again.

I still get the error above.

Now I am building again without the MATLAB flag, and it is going fine.

varunagrawal, 2019-07-10:
Okay I can reproduce this. I'll see what the issue is.

jlblancoc, 2019-07-10:
@varunagrawal Let me know if you suspect it's related to any of the recent changes to cmake. 

I don't have any Linux partition with MATLAB installed (for some reason, always preferred Windows for matlab), but if needed to debug this, could give it a try.

varunagrawal, 2019-07-11:
@jlblancoc seems like [this](https://github.com/borglab/gtsam/commit/7f43054c37930261ddd88716e79dba896ef02d1a) commit broke something. I am able to build the MATLAB wrapper on commits before this.

I'm looking into it some more to try and isolate the issue.

varunagrawal, 2019-07-11:
I found the problem.

In `gtsam/CMakeLists.txt` you are setting the project again via `project(gtsam LANGUAGES CXX)`. The project name should be `GTSAM` aka uppercase, which is why the check at L102 of `GtsamMatlabWrap.cmake` is failing (it is checking for `GTSAM` but the project name has been set to `gtsam` (lowercase), and hence it goes to the else condition and fails because `GTSAM_INCLUDE_DIR` is never set explicitly in the cmake files.

The fix should be super simple: Update `gtsam/CMakeLists.txt` to set the project to `GTSAM` so that we are consistent, and then update `GtsamMatlabWrap.cmake` to check for lowercase or uppercase always.

@izzys thanks again for submitting the issue. We should have this resolved by tomorrow.

varunagrawal, 2019-07-11:
@dellaert you should take a look at this issue. This is why we really need to figure out CI for Matlab. Maybe contact Mathworks about this?

jlblancoc, 2019-07-11:
Thanks for investigating it!

But please, don't change it to GTSAM, it's like that for a reason (undoing it will break Windows builds). 

Why not updating GtsamMatlabWrap to check for lowercase only? Can you try it? Thanks

cntaylor, 2019-07-11:
I had this same problem.  I fixed this (see github cntaylor/gtsam, about to submit a pull request...) by modifying line 102 in GtsamMatlabWrap to check for "gtsam" rather than "GTSAM"

PR #85 (2019-07-19): (WIP) Changes to get gtsam to compile in Windows
Follow up of #84 . Fixes #82 . 

To-do: 
- [x] remove whitespace changes, 
- [x] remove the definition of NOMINMAX in the gtsam .h file, and add it via cmake definitions, which is a much cleaner way.

---
This change is [[image]](https://reviewable.io/reviews/borglab/gtsam/85)
closed 2019-07-19 by jlblancoc