borglab/gtsam #86: Linking errors on MSVC (2017 Community)

cntaylor, 2019-07-11:
## Description
In MSVC 2017 (Community), including the gtsam library causes linking errors.  Specifically, gstam_matlab_wrapper, testLieMatrix, testLieVector, testLieScalar, and testMatrix does not finish compiling.  During the linking phase, 100s of errors are generated, but they all have to do with an "unresolved external symbol __declspec(dllimport)" and it talks about Eigen.  An example error:

```
 Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::LieVector::LieVector(class Eigen::Matrix<double,-1,1,0,-1,1> const &)" (__imp_??0LieVector@gtsam@@QEAA@AEBV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@@Z) referenced in function "public: virtual void __cdecl LieVectorInvariantsTest::run(class TestResult &)" (?run@LieVectorInvariantsTest@@UEAAXAEAVTestResult@@@Z)	check_base_program	C:\AFIT\lib\gtsam\build\gtsam\base\tests\testLieVector.obj		

```

## Steps to reproduce

1. Use the branch msvc-fixes
2. Run Cmake
  * Some switches of possible importance:
    * GTSAM_INSTALL_MATLAB_TOOLBOX is on
    * GTSAM_BUILD_TESTS is on
    * GTSAM_BUILD_WRAP is on
    * GTSAM_ALLOW_DEPRECATED... is off
    * GTSAM_BUILD_UNSTABLE is off
    * MEX directory and executable location are set appropriately
    * GTSAM_USE_SYSTEM_EIGEN is off (I don't have Eigen installed separately, so it can't be a conflict)
    * GTSAM_WITH_TBB is off
    * GTSAM_MEX_BUILD_STATIC_MODULE is off (documentation says it must be off for Windows!)
3. Open the solution file and try to compile.  gtsam compiles properly. Unit tests and gtsam_matlab_wrapper does not. 

## Expected behavior
Code completes compiling without errors

## Environment
Windows 10, Visual Studio 2017 Community, Boost 1.67.  Errors appear in DEBUG or RELEASE.

## Additional information
I am not sure, but I think it has to do with something like the problem described on [this webpage](https://community.kde.org/Windows/Build/Common_problems).  Because Matrix is not declaring the size of the Matrix up front, then you have a not fully defined type being used as an interface to a dll.  That is my best guess though and I have never dug into how dlls work on Windows.

cntaylor, 2019-07-12:
I think I have fixed this issue ... I will push up my changes after I am done testing and then hopefully close this issue, but wanted to get some thoughts/questions out there.

The primary source of error was LieVector, LieScalar, and LieMatrix.  Because there are no .cpp files associated with these classes, the test and wrapper functions complained during compile as they try to call the functions in these header files.  I can fix all the compilation problems (everything builds, but I haven't tried the generated MATLAB toolbox yet ... waiting on a re-compile) by simply removing these three files from all the CMakeLists.txt files and from any file that tries to call them.  BUT, it disturbs me that everything compiles in Linux, even with the headers for these classes but not the .cpp files.  Why does Linux compile without LieVector, LieScalar, or LieMatrix.cpp (and pass all the tests, including testLieMatrix) when there are no .cpp files defining what those classes are?

Thoughts?

cntaylor, 2019-07-13:
Pull request #87 closes this issue...

dellaert, 2019-07-13:
PR #87 is not the right fix, see my comment there...

cntaylor, 2019-07-19:
Fixed with most recent PR #87 that is in the develop branch now.

closed 2019-07-19 by cntaylor