borglab/gtsam #90: Linux Packaging

acxz, 2019-07-21:
## Feature

It would be great to have gtsam as a GNU/Linux package.

## Motivation

Being able to `sudo apt install gtsam` will make it easy to install and develop on the gtsam library. Building from source does take a significant amount of time and is tedious. Having a precompiled package on for example, https://launchpad.net/, will speed this up and allow the development group to be more in control of the release versions available to the users.

## Pitch

Have packages for widely available GNU/Linux distributions, such as Debian, Ubuntu, etc.

## Alternatives

N/A

## Additional context

N/A

jlblancoc, 2019-07-22:
Hi,
Do you mean... [this](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/gtsam-develop)? ;-)

acxz, 2019-07-22:
Ah I see, it seems like I did not search hard enough. Maybe having the Linux packages available on the gtsam website (under Getting started or Build) would be a good idea for visibility.

In addition to the nightly release builds for Ubuntu builds, it would be nice to have the latest stable versions on the PPA. Also as you mentioned the current packaging does not conform to the Linux Filesystem Hierarchy Standard, which would be necessary for a Debian package instead of a community build on Launchpad.

So I guess these 3 things would be nice to have:
1. Visibility of Linux Packages on GTSAM website
2. Stable Releases on Launchpad
3. Conform Debian-based packages to the Linux FHS

dellaert, 2019-07-30:
I fully support you both providing all of the above :-) GTSAM website is also in a repo and open to PR's, just Github Pages based...

varunagrawal, 2019-07-30:
Maybe consider a [snap](https://snapcraft.io/) instead of a deb package?

dellaert, 2019-07-30:
Why?

varunagrawal, 2019-07-30:
Snaps are universal so they are not restricted to any particular distribution, unlike deb packages which are restricted to Debian based distributions.

As a side, I am personally of the opinion that Linux packaging is a mess due to its being so opinionated. I prefer having a Homebrew based installation especially since Homebrew now supports Linux and Mac by default.

acxz, 2019-08-03:
I have my own reservations on snaps and personally prefer traditional Linux packaging. Also in regards to Homebrew, I believe [vcpkg](https://github.com/Microsoft/vcpkg) is better in this aspect since it also supports Windows.

All in all tho, I am of the opinion that software developers, such as the GTSAM group, should not worry about packaging at all. The onus is on the respective community (whether it be the homebrew community or individual Linux distro communities) to go from a source install to the package. If there is a need is said community, the community will create the package. It is just that packaging helps raise visibility of software and a package meshes better with an operating system than source installs. This is usually the reason why development groups will at least officially support one form of packaging (preferably a popular one such as Debian, whose software package repos are used by many Debian derivatives including Ubuntu).

varunagrawal, 2019-09-18:
@jlblancoc I believe you've already done this? Can we close it?

dellaert, 2019-09-18:
@jlblancoc if you bieve this has been addressed we'll let you close.

jlblancoc, 2019-09-18:
Well... I would say it's sort of done, in the practical sense, since anyone can quickly install gtsam right now from the PPA in seconds for any architecture (arm64, ...). 

Now, it remains an open issue the proper packaging for, for example, Debian (Ubuntu), which would require more work. But normally these things are handled by different people than those working "upstream", so... yes, I'll close this one, and if proper packaging requires changes upstream someday, let new issues to be open for that.

closed 2019-09-18 by jlblancoc