borglab/gtsam #94: "make doc" does not work

vegnesh, 2019-07-29:
Typing "make doc" in the build directory does not work  (instruction given in INSTALL.md)

I generated a build directory in doc folder and did cmake .. and make and that gave the following error
/bin/sh: 1:/path_to_doc_dir/build/Doxyfile: Permission denied

Is there anything I am missing out?

dellaert, 2019-08-01:
Is doxygen installed? cmake checks that..

vegnesh, 2019-08-01:
I have Doxygen installed, however there is no target named doc in the makefile generated using CMake.Is there a flag that needs to be enable while generating the makefile using CMake to generate documentation?

dellaert, 2019-08-01:
This par of the cmake code is quite old, and I'm no expert. It could be, or could be broken. If you do some digging and find out (or whomever is watching this) it would be highly appreciated. Probably a simple issue...

acxz, 2019-08-03:
@vegnesh 

Note you need to call the `CMakeLists.txt` of the entire project, not the one in the `doc/` folder to create all the `make` targets like `doc`. The project wide `CMakeLists.txt` would be called by `cmake ../..` if you have a build folder inside the `doc/` folder, instead of `cmake ..` which is what you did.

I just tried the following on my system:
```
git clone https://github.com/borglab/gtsam
cd gtsam/doc
mkdir build
cd build
cmake ../..
make doc
```
and that generated the docs properly.
I get the exact error based on the procedure you mentioned, which is fine since that process is not expected to work.

jlblancoc, 2019-08-04:
To avoid users falling into these errors (which seem common when starting with cmake?), there is a solution: introducing an `include(...)` to the top of each CMakeLists.txt file except the root one, to do a check like [this one](https://github.com/MRPT/mrpt/blob/master/cmakemodules/AssureCMakeRootFile.cmake). 

Pros: avoids this kind of error.
Cons: it's "visually unpleasant" to see that include polluting the cmake files at all subdirectories! ;-)

Is it worth? I don't know :-)

vegnesh, 2019-08-05:
@acxz Your method works. I had done what you had suggested previously. However, at that point doxygen was not installed in my machine and hence there was no target  for 'doc' generated in the Makefile. Now, it seems to work. Sorry for creating confusion

closed 2019-08-05 by vegnesh