borglab/gtsam #95: GTSAM related compile error with BLAM

BMannell, 2019-07-30:
This issue is centered around the [blam](https://github.com/erik-nelson/blam) library. I get the following error when trying to build blam.

```
In file included from /home/ben/Documents/blam/internal/src/laser_loop_closure/include/laser_loop_closure/LaserLoopClosure.h:45:0,
                 from /home/ben/Documents/blam/internal/src/laser_loop_closure/src/LaserLoopClosure.cc:37:
/usr/local/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^
```

Is there a way to enforce GTSAM to use a specific version of Eigen? 
GTSAM seems to come packaged with a version of Eigen, is there some way to get it to use the previously installed version? 
How can I tell which version is being used in order to align the version of the other package? 

OS: Ubuntu 16.04 LTS 
Latest versions of each library from github repos.

jlblancoc, 2019-07-30:
Try setting `GTSAM_USE_SYSTEM_EIGEN=ON`

BMannell, 2019-07-30:
@jlblancoc Thank you!

closed 2019-07-31 by BMannell