borglab/gtsam #96: IndeterminantLinearSystemException when computing marginals

mbrossar, 2019-08-01:
I search to compute covariance marginals from a pose-graph that does not have any loop closure.

This happen even after slightly modifying Pose3SLAMExample_g2o as follow:
1-include marginal computation in examples/Pose3SLAMExample_g2o.cpp [Pose3SLAMExample_g2o.cpp](https://github.com/borglab/gtsam/files/3456031/Pose3SLAMExample_g2o.txt)
2-remove loop-closure in examples/Data/pose3example.txt [pose3example.txt](https://github.com/borglab/gtsam/files/3456027/pose3example.txt)

The optimization is done, but marginals computation is not done. Error message is:
terminate called after throwing an instance of 
'gtsam::IndeterminantLinearSystemException' what():  
Indeterminant linear system detected while working near variable 4 (Symbol: 4).

With prior of the first pose, I do not see why system could be underdetermined.

dellaert, 2019-08-01:
Hmm, this would make it similar to LocalizationExample. If you help me by creating a tiny PR that adds this example (and the file) as Pose3LocalizationExample.cpp, I'll try to repro and diagnose. If we fix it, it can remain as a new example :-)

mbrossar, 2019-08-02:
Thank you, the PR has just be done.

closed 2020-04-04 by mbrossar