borglab/gtsam #98: Can't debug into gtsam source file

huanhexiao, 2019-08-04:
i compiled my own source files together with gtsam, i put my files in an independent fold.
i can't debug into gtsam source file.
My develop info:
Sys: Ubuntu 18.04
Boost: 1.65.1
Eigen:GTSAM_USE_SYSTEM_EIGEN( 3.3.7)

help please!

huanhexiao, 2019-08-04:
I use Vscode

varunagrawal, 2019-08-04:
Are you using the debugger in VS Code? Have you set it up correctly?

dellaert, 2019-08-04:
This is not a GTSAM issue but a request for technical support. FWIW I have no issues setting up a debug config in VS code, but it's not super -straightforward. read the manuals carefully. If that does not help (and only then), please post on the GTSAM users group at https://groups.google.com/forum/#!forum/gtsam-users

closed 2019-08-04 by dellaert

huanhexiao, 2019-08-05:
> Are you using the debugger in VS Code? Have you set it up correctly?

yes, it's in vscode. now i've found the reason, i should specify CMAKE_BUILD_TYPE before everything.
thank u very much!

huanhexiao, 2019-08-05:
> This is not a GTSAM issue but a request for technical support. FWIW I have no issues setting up a debug config in VS code, but it's not super -straightforward. read the manuals carefully. If that [...]
Of course, thank u frank