borglab/gtsam #99: Missing EIGEN_MAKE_ALIGNED_OPERATOR_NEW in some classes

thorstink, 2019-08-05:
## Description

I am having non-consistent behaviour with using a memory-sanitizer (```-fsanitize=address```). I get different results for instance depending on when I run it locally on my machine versus running the same in a Docker... I suspect it has to do with the use of the ```EIGEN_MAKE_ALIGNED_OPERATOR_NEW```-flag.

Eigen has a special define that should be used in classes that have Eigen-types as member variables: ```EIGEN_MAKE_ALIGNED_OPERATOR_NEW```. Many GTSAM-classes have this; e.g. ```CombinedImuFactor.h``` and ```ImuBias.h```, however, many do not have this flag even though they have Eigen-members: ```ImuFactor.h``` for instance. Some do not have it but inherit it from their parent class, e.g. ```ManifoldPreintegration.h``` does not have it, but its base-class ```PreintegrationBase.h``` does. And  ```TangenPreintegration.h``` does have it (even though its base-class also has it)..  Noteworthy is that none of the noisemodels use it.

## Steps to reproduce
I do not have a minimal working example yet to share. If I build everything (gtsam & project) with  ```Debug``` it is fine. However, using ```Release``` I get something like:
```bash
1: ==15494==ERROR: AddressSanitizer: attempting free on address which was not malloc()-ed: 0x607000000b20 in thread T0
1:     #0 0x7f02eeb3d7b8 in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xde7b8)
1:     #1 0x7f02ee09a6ae in Eigen::internal::compute_inverse<Eigen::Matrix<double, -1, -1, 0, -1, -1>, Eigen::Matrix<double, -1, -1, 0, -1, -1>, -1>::run(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&) (/usr/local/lib/libgtsam.so.4+0x25d6ae)
1:     #2 0x7f02ee09675e in gtsam::noiseModel::Gaussian::Covariance(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, bool) (/usr/local/lib/libgtsam.so.4+0x25975e)
1:     #3 0x7f02ee14b581 in gtsam::CombinedImuFactor::CombinedImuFactor(unsigned long, unsigned long, unsigned long, unsigned long, unsigned long, unsigned long, gtsam::PreintegratedCombinedMeasurements const&) (/usr/local/lib/libgtsam.so.4+0x30e581)

```
## Expected behavior
No problems with the sanitizer in release.

## Questions/next steps
I expect it works in Debug because it (probably?) prevents Eigen from doing its fancy alignment-moves..

Is the inconsistent use of ```EIGEN_MAKE_ALIGNED_OPERATOR_NEW``` with a purpose throughout GTSAM? If not, I am ok with working through as much of GTSAM as I can and add many of these Eigen-flags where appropriate. But I'd like some of your thoughts before I start million-changes approach.

 Any other ideas or clues?

dellaert, 2019-08-05:
Just a quick note: not every class needs the directive: only the ones with specific alignments. Point2 needs it, Point3 does not, for example. So blanket addition is not the answer. That being said, it would be good to get to the bottom of the error above. It seems to involve a dynamically sized matrix, so are we sure it's alignment-related?

thorstink, 2019-08-06:
Oh, yes. I knew that about the directive. I am not sure it is alignment related, but it was one of my first guesses (because of some other sanitizer input I saw). I haven't structured debugging this particular error properly yet. For the upcomming few days I have to put focus on something else, but after that I'll attempt to make a minimum working example and get a bit more clear on what is happening. 

For now, a (maybe related) note, what also pop upped in the sanitizer is a similiar complaint about Lago.
In ```Lago.cpp``` there are globally static const noisemodels. These sometimes also trigger the memory sanitizer to complain (even though I am not using Lago in my application, but it's a globally static define.. so maybe we should change that)...

I'll report back once I know more.

thorstink, 2019-08-29:
I’m back from holiday and will dig into this again!

dellaert, 2019-09-07:
Did you find out anything? Make sure that if march=native flag is used in GTSAm, it is also used in project you link with. That's a common issue.

thorstink, 2019-09-09:
It's still on my list (getting to it..)

I did already rule out march-flags..

thorstink, 2019-09-23:
fyi;

My problem is that I don't have a lot of time to make a isolated example, but I'll do my best.... I'm trying to trim down the example to something I can share.. So far I excluded the FixedLag smoothner but no luck other then that..

I'll also reinvestigate the compiler-architecture flagging.. because I still don't trust them (we have quite some confusing stuff with arm-flagging -.-). 

```bash
1: Test timeout computed to be: 1500
1: =================================================================
1: ==3907==ERROR: AddressSanitizer: attempting free on address which was not malloc()-ed: 0x603000000350 in thread T0
1:     #0 0x7f3eb224d7b8 in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xde7b8)
1:     #1 0x7f3eb18a669a in Eigen::internal::aligned_free(void*) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:177
1:     #2 0x7f3eb18a669a in void Eigen::internal::conditional_aligned_free<true>(void*) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:230
1:     #3 0x7f3eb18a669a in void Eigen::internal::conditional_aligned_delete_auto<double, true>(double*, unsigned long) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:416
1:     #4 0x7f3eb18a669a in Eigen::DenseStorage<double, -1, -1, 1, 0>::~DenseStorage() /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:542
1:     #5 0x7f3eb18a669a in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::~PlainObjectBase() /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:98
1:     #6 0x7f3eb18a669a in Eigen::Matrix<double, -1, 1, 0, -1, 1>::~Matrix() /usr/include/eigen3/Eigen/src/Core/Matrix.h:178
1:     #7 0x7f3eb18a669a in gtsam::noiseModel::Constrained::MixedSigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&) /tmp/gtsam-4.0.0/gtsam/linear/NoiseModel.h:419
1:     #8 0x7f3eb18a669a in gtsam::noiseModel::Diagonal::Sigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, bool) /tmp/gtsam-4.0.0/gtsam/linear/NoiseModel.cpp:271
1:     #9 0x7f3eb1755eae in __static_initialization_and_destruction_0 /tmp/gtsam-4.0.0/gtsam/slam/lago.cpp:37
1:     #10 0x7f3eb1755eae in _GLOBAL__sub_I_lago.cpp /tmp/gtsam-4.0.0/gtsam/slam/lago.cpp:399
1:     #11 0x7f3eb313a732  (/lib64/ld-linux-x86-64.so.2+0x10732)
1:     #12 0x7f3eb312b0c9  (/lib64/ld-linux-x86-64.so.2+0x10c9)
1: 
1: 0x603000000350 is located 16 bytes inside of 24-byte region [0x603000000340,0x603000000358)
1: allocated by thread T0 here:
1:     #0 0x7f3eb224db50 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb50)
1:     #1 0x55b4c683944b in Eigen::internal::handmade_aligned_malloc(unsigned long) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:88
1:     #2 0x55b4c683958d in Eigen::internal::aligned_malloc(unsigned long) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:164
1:     #3 0x55b4c68428eb in void* Eigen::internal::conditional_aligned_malloc<true>(unsigned long) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:214
1:     #4 0x55b4c687a4f5 in double* Eigen::internal::conditional_aligned_new_auto<double, true>(unsigned long) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:374
1:     #5 0x55b4c6889029 in Eigen::DenseStorage<double, -1, -1, 1, 0>::resize(long, long, long) /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:557
1:     #6 0x55b4c687aaec in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::resize(long, long) /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:293
1:     #7 0x55b4c6865630 in void Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::resizeLike<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > >(Eigen::EigenBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&) /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:375
1:     #8 0x55b4c68521c8 in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::PlainObjectBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > >(Eigen::DenseBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&) /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:536
1:     #9 0x7f3eb18a6682 in Eigen::Matrix<double, -1, 1, 0, -1, 1>::Matrix<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > >(Eigen::EigenBase<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&) /usr/include/eigen3/Eigen/src/Core/Matrix.h:379
1:     #10 0x7f3eb18a6682 in gtsam::noiseModel::Constrained::MixedSigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&) /tmp/gtsam-4.0.0/gtsam/linear/NoiseModel.h:419
1:     #11 0x7f3eb18a6682 in gtsam::noiseModel::Diagonal::Sigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, bool) /tmp/gtsam-4.0.0/gtsam/linear/NoiseModel.cpp:271
1: 
1: SUMMARY: AddressSanitizer: bad-free (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xde7b8) in __interceptor_free
1: ==3907==ABORTING
1/1 Test #1: odom .............................***Failed    0.86 sec
```

ProfFan, 2019-09-23:
It is somewhat related to translation units. `handmade_aligned_malloc` should be freed with `handmade_aligned_free`.

Can you try removing the system Eigen and use the GTSAM included one instead?

thorstink, 2019-09-24:
Will try to do it today.. have to rebuild an annoying docker image -.-

ProfFan, 2019-09-27:
Note that you do NOT need to turn off march=native. Have you added a apt-get remove eigen in the Dockerfile?

thorstink, 2019-10-02:
I removed eigen from the docker-file and used gtsam's eigen.. no luck :/ 

Also, I never had march=native turned on.. but march=native turned on 'fixes' it... except for a TBB-error.. but I'll now try to see what happens if I update tbb to 2019.

```no-lang

1: Test command: /mnt/salsa_ws/unittest/build/odometry/odometry_test
1: Test timeout computed to be: 1500
1: ===============================================================================
1: All tests passed (406 assertions in 3 test cases)
1: 
1:                                     
1: =================================================================
1: ==3197==ERROR: LeakSanitizer: detected memory leaks
1: 
1: Direct leak of 3096 byte(s) in 3 object(s) allocated from:
1:     #0 0x7f9b019e6618 in operator new[](unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe0618)
1:     #1 0x7f9b00b75752  (/usr/lib/x86_64-linux-gnu/libtbb.so.2+0x21752)
1: 
1: SUMMARY: AddressSanitizer: 3096 byte(s) leaked in 3 allocation(s).
1/1 Test #1: odom .............................***Failed    0.48 sec
```

ProfFan, 2019-12-09:
Closing as resolved. Thank you for the report!

closed 2019-12-09 by ProfFan

fhill122, 2021-03-25:
Hi in my test, disabling march=native does solve this issue, BUT this requires any other libraries to be compiled with march=native as well.  When the depency list is long, this could be tedious...
A simple case could be the libpcl installed from apt in ubuntu: despite "disabling tbb, enabling march=native", compiling with pcl would still produces sanitizer errors.

ProfFan, 2021-03-25:
@fhill122 Please open a new issue with *all* your logs and reproduction steps according to the issue template. If you think it's related to this issue, link to this issue in your issue. Thanks :)

OznOg, 2022-08-03:
I had the same error and here my analysis:
Eigen (in Memory.h) does things like:
```
EIGEN_DEVICE_FUNC inline void aligned_free(void *ptr) 
{ 
 #if (EIGEN_DEFAULT_ALIGN_BYTES==0) || EIGEN_MALLOC_ALREADY_ALIGNED 
   std::free(ptr); 
 #else 
   handmade_aligned_free(ptr); 
 #endif 
}
```
the `EIGEN_MALLOC_ALREADY_ALIGNED` is defined as:

```
#if (...) && ! defined( __SANITIZE_ADDRESS__ ) && (...)
 #define EIGEN_GLIBC_MALLOC_ALREADY_ALIGNED 1 
#else 
 #define EIGEN_GLIBC_MALLOC_ALREADY_ALIGNED 0 
#endif
```

So the header detects if it is used while compiling with sanitizer, and changes the allocation policy in this case.

This is perfectly fine because eigen is header only thus `EIGEN_MALLOC_ALREADY_ALIGNED` cannot change value.

The issue come from the fact that GTsam is not header only but pulls eigen headers (thus Memory.h) thru its own headers. This lead to have a core GTsam.so that was compiled with `EIGEN_GLIBC_MALLOC_ALREADY_ALIGNED` set to false and later on, when compile the code that use gtsam using sanitizer, files are then compiles with `EIGEN_GLIBC_MALLOC_ALREADY_ALIGNED` set to true, leading at runtime to the error.

Hard to propose a workaround...