borglab/gtsam #100: Travis not triggered

dellaert, 2019-08-07:
## Description

Travis request is not handled correctly in release branch.
Travis says: Not found: borglab/gtsam/.travis.yml@942a4d852bc74f2d

## Steps to reproduce

1. push release/3.2.3 (where I added travis.yml and travis.sh)

## Expected behavior

Travis does CI

## Environment

Pushed from Mac, but irrelevant?

dellaert, 2019-08-07:
@jlblancoc any idea?

jlblancoc, 2019-08-08:
The option "build pushed branches" was disabled, and it allowed me to enable it so... in theory, the next commit should trigger a build!

dellaert, 2019-08-08:
Hmmmm, no :-(

varunagrawal, 2019-08-08:
Perhaps try adding the safelist directive to `.travis.yml`?

https://docs.travis-ci.com/user/customizing-the-build/#safelisting-or-blocklisting-branches

dellaert, 2019-08-08:
Turns out I missed the dot in the travis files....

closed 2019-08-08 by dellaert