borglab/gtsam #101: SmartProjectionFactor inconsistency

dellaert, 2019-08-08:
## Description

Back in December, Nikolaus Mitchell (@mitch2na ??) ran into issues with the `SmartProjectionFactor`. The bug report is here: https://bitbucket.org/gtborg/gtsam/issues/411/smartposeprojfactor-error-with-sensor

I started looking into it but then got side-tracked. I have now come back to it and I feel that giving a sensor offset `body_P_sensor` to the `SmartProjectionFactor` does not make sense, and I think the current constructor should be deprecated to prevent confusion. Indeed, It does not make sense to optimize for a camera where the pose would not be the actual pose of the camera. An unfortunate consequence of deprecating this constructor means that we cannot optimize for *individually varying* camera calibrations when the camera is offset from the body pose. That would need a new factor with (body) pose and calibration as variables. However, that use case is unlikely: when a global offset is know, calibration is typically known.

## Proposed remedy

Deprecate the constructor

mikesheffler, 2019-08-21:
>  I have now come back to it and I feel that giving a sensor offset body_P_sensor to the SmartProjectionFactor does not make sense, and I think the current constructor should be deprecated to prevent confusion.

Currently, the `SmartProjectionPoseFactor` constructor just forwards `body_P_sensor` to the `SmartProjectionFactor` constructor. Are you proposing to keep `body_P_sensor` to `SmartProjectionPoseFactor`, or get rid of it entirely?

> Indeed, It does not make sense to optimize for a camera where the pose would not be the actual pose of the camera. 

As in, a non-trivial `body_P_sensor`? I was thinking of the case where you have ...

 *  Single monocular camera going forward in time
 * There is a body <-> sensor offset
 *  The offset is non-moving (as in, 'no optimization for body_P_sensor`')
 *  Fixed camera intrinsics

Would that be precluded with what you are thinking?

dellaert, 2019-08-21:
@mikesheffler 
I had already created  a branch for this but apparently forgot to do a PR :-) Probably because I was still trying to decide for myself whether it makes sense or not. Anyway, I added you as a collaborator and as a reviewer, take a look at it.

mikesheffler, 2019-08-26:
Ooooookay. I misunderstood (reading here, through the issues) what you were trying to do. I think now that I might have misunderstood what Nikolaus Mitchell wanted in his bug report, as well.

Now that I've looked at  #105 , I get it. I agree with your statement about how the transform doesn't make sense at `SmartProjectionFactor` given the inheritance chain.

I think your change makes sense. I'll comment more over on the PR itself.

dellaert, 2019-09-07:
Fixed in PR #105

closed 2019-09-07 by dellaert