borglab/gtsam #103: The ImuFactorExample2 is throwing an exception on second isam update

biggibig, 2019-08-08:
Windows 10 64bit
Visual Studio 2017 15.4.5
GTSAM project generated using CMAKE no TBB no MKL, using precompiled boost_1_70

The second time line 126 isam.update(newgraph, initialEstimate); gets called, the program throws line 282 in Values-inl.h throw ValuesIncorrectType(j, typeid(*pointer), typeid(ValueType));

This appears to happen on the second call to the operator.

jlblancoc, 2019-08-14:
The latest git develop version of the example runs cleanly (even under Valgrind) when built in GCC. 

What version of gtsam are you using?

biggibig, 2019-08-19:
I just pulled down "develop" on Aug 08 2019.

The interesting thing is that ImuFactorExample seems to be working fine. I can even make the isam2 updater and use it. Trying to go the other way and try to use the LM updater in ImuFactorExample2 does not work, the same exception gets thrown.

dellaert, 2019-08-19:
That's weird. Seems windows-related, maybe compiler that is more anal about RTTI. Could you investigate in a bit more detail what exactly the issue is?

biggibig, 2019-08-20:
If I figure out the exact thing that breaks it, I'll let you know.

ProfFan, 2019-09-23:
#121

biggibig, 2019-10-17:
I have a question about intended behavior of some of this stuff.

Let's say you have two 6DOF locations 2 priors, 1 binary factor. A situation where all values and factors have the same type obviously works.
initial(0) gets a Pose3; priorFactor(0) gets a Pose3;
initial(1) gets a Pose3, priorFactor(1) gets a Pose3;
betweenFactor(0,1) gets a Pose3;

You can change the Pose3 to a Rot3 for all of them and that still works.

Should I be able to mix and match types on the factors? Say by changing the second prior to a Rot3. As in

initial(0) gets Pose3; priorFactor(0) gets Pose3
initial(1) gets Pose3; priorFactor(1) gets Rot3
betweenFactor(0,1) gets a Pose3;

Is this scenario supposed to work? Because it's throwing the same exception for me.

dellaert, 2019-10-17:
No, a prior factor needs to match the type of the variable. It is possible to have factors that only provide information about part of the variable (e.g., AttitudeFactor is an example), but that is not a “prior” on the variable in that case.

Could it be that this issue is not a GTSAM bug but rather the result of not being quite familiar yet with GTSAM and factor graphs? if so, please close the issue, read more (e.g. 2017 booklet by myself and Michael Kaess), and ask for help on Google Group - not via the GitHub issue system.

biggibig, 2019-10-17:
Thanks that's exactly what I suspected.

The original issue is that the ImuFactorExample2 is throwing an exception on update after the IMU factor is added, without any modifications to the code on my part.  Hence I needed to double check that the exception thrown by my simple dummy problem was in fact supposed to be thrown.

ProfFan, 2019-12-09:
Closing due to inactivity.

closed 2019-12-09 by ProfFan