borglab/gtsam #104: gtsam with Eigen when catkin_make errors

YangSiri, 2019-08-10:
Im using ros(kinetic), after installed ros, i build gtsam from source and it went well. nothing happens. But when i using catkin_make to build a workspace, Error comes. 

Error description: 

  In file included from /usr/local/include/gtsam/base/Matrix.h:26:0,
                 from /usr/local/include/gtsam/base/Manifold.h:22,
                 from /usr/local/include/gtsam/base/Lie.h:25,
                 from /usr/local/include/gtsam/base/VectorSpace.h:11,
                 from /usr/local/include/gtsam/geometry/Point2.h:20,
                 from /usr/local/include/gtsam/geometry/Unit3.h:23,
                 from /usr/local/include/gtsam/geometry/Rot3.h:25,
                 from /home/joe/workspace/catkin_ws/src/LeGO-LOAM/LeGO-LOAM/src/mapOptmization.cpp:36:
/usr/local/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(

#end description

Would someone help me ? Thanks very much!

varunagrawal, 2019-08-10:
The error message gives you the problem: ROS/catkin seems to be using a different version of Eigen than the one you used to compile GTSAM with. You may have to manually specify the Eigen path using cmake during compilation.

YangSiri, 2019-08-10:
Thanks,  what you mean is there are two kinds of eigen in my system? But i already rebuilt them two, why didnt it work?

varunagrawal, 2019-08-11:
There is one version of Eigen that possibly came with ROS or was already on your system, and there seems to be another one, which is most likely what you have installed manually.

This is a support question and not a GTSAM problem. I highly recommend asking this doubt on the Google group.

closed 2019-08-11 by varunagrawal

hello-sea, 2019-08-21:
This helps me:
```shell
mkdir build && cd build
cmake -DGTSAM_USE_SYSTEM_EIGEN=ON ..
sudo make install
```
@YangSiri

YangSiri, 2019-08-21:
@hello-sea 

Thanks, but i tried. it doesnt work...

TixiaoShan, 2019-09-21:
> This helps me:
> 
> ```shell
> mkdir build && cd build
> cmake -DGTSAM_USE_SYSTEM_EIGEN=ON ..
> sudo make install
> ```
> 
> @YangSiri

This solved my problem without changing any of my code

YangSiri, 2019-09-21:
@hello-sea  @TixiaoShan @varunagrawal 
Thank you all for the attention, I've found the problem: i was not using the specific version of gtsam you (@TixiaoShan ) mentioned in LeGO-loam. Instead I was using the newest version and after i changed it. It works now. 
But it seems that i cannot use this gtsam in other projects. Such like when i add gtsam in the CMakelists and AS LONG AS #include some gtsam headers in my cpp, it goes like "Process finished with exit code 134 (interrupted by signal 6: SIGABRT)"  even though i reinstalled it some times. Some headers are fine, but #include <gtsam/slam/BetweenFactor.h> doesnt !

Do you have any ideas about this ? I would be grateful.

TixiaoShan, 2019-09-21:
@YangSiri 
I am using the latest GTSAM for LeGO-LOAM, it seems to be working fine. Sorry I have no insight for your problem.

YangSiri, 2019-09-21:
@TixiaoShan  
You do ?  But i had to download the gtsam you provided. And it would work. Maybe i was wrong. Sorry, But why cant i use it in other projects ? Im really confused.

TixiaoShan, 2019-09-21:
@YangSiri 
I think something is changed in GTSAM between the alpha version and the current version. Honestly, I don't know the real reason behind this.

YangSiri, 2019-09-21:
@TixiaoShan 
Yes, thank you anyway. I think you had done a great work !

MohamedAfifii, 2019-11-25:
@YangSiri 

This problem can also rise if you have different Eigen versions on your system. For example I faced this problem when GTSAM used Eigen from `/usr/local/include` when I specified `-DGTSAM_USE_SYSTEM_EIGEN=ON`, while the other library that I was trying to build used Eigen from `/usr/include`.

YangSiri, 2019-11-25:
Hi， @MohamedAfifii  thanks
Now this problem is gone somehow. But i cant use gtsam in my own project. 

So you were saying we'd better avoid using two different version of Eigen in one project(CMakeList)? There seems to be two version of gtsam, how can i remove it?

Angit16, 2020-07-06:
@YangSiri 
cat install_manifest.txt | xargs echo sudo rm | sh 
Please check if it helps.