borglab/gtsam #114: IndeterminantLinearSystemException for big numbers

ClementLeBihan, 2019-09-17:
# Description

Hi ! 

I'm trying to use gtsam with big values for long car trajectory. The program thrown with this error 
`terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
5 (Symbol: 5).` when using big numbers.

To show it's not a underconstrained variables problem, I made some change to Pose2SLAMExample to get 2 different examples with one that works and the other that crash. 
Here are the 2 part of code : 

1.  THE CODE THAT WORKS FINE : 
`  noiseModel::Diagonal::shared_ptr priorNoise = noiseModel::Diagonal::Sigmas(Vector3(3e-6, 3e-6, 1e-3));
  graph.emplace_shared<PriorFactor<Pose2> >(1, Pose2(0, 0, 0), priorNoise);

  // For simplicity, we will use the same noise model for odometry and loop closures
  noiseModel::Diagonal::shared_ptr model = noiseModel::Diagonal::Sigmas(Vector3(2e-6, 2e-6, 1e-3));

  // 2b. Add odometry factors
  // Create odometry (Between) factors between consecutive poses
  graph.emplace_shared<BetweenFactor<Pose2> >(1, 2, Pose2(20000, 0, 0     ), model);
  graph.emplace_shared<BetweenFactor<Pose2> >(2, 3, Pose2(20000, 0, M_PI_2), model);
  graph.emplace_shared<BetweenFactor<Pose2> >(3, 4, Pose2(20000, 0, M_PI_2), model);
  graph.emplace_shared<BetweenFactor<Pose2> >(4, 5, Pose2(20000, 0, M_PI_2), model);

  // 2c. Add the loop closure constraint
  // This factor encodes the fact that we have returned to the same pose. In real systems,
  // these constraints may be identified in many ways, such as appearance-based techniques
  // with camera images. We will use another Between Factor to enforce this constraint:
  graph.emplace_shared<BetweenFactor<Pose2> >(5, 2, Pose2(20000, 0, M_PI_2), model);
  graph.print("\nFactor Graph:\n"); // print

  // 3. Create the data structure to hold the initialEstimate estimate to the solution
  // For illustrative purposes, these have been deliberately set to incorrect values
  Values initialEstimate;
  initialEstimate.insert(1, Pose2(50000,     0,  0.2   ));
  initialEstimate.insert(2, Pose2(23000,  1000, -0.2   ));
  initialEstimate.insert(3, Pose2(41000,  1000,  M_PI_2));
  initialEstimate.insert(4, Pose2(40000, 20000,  M_PI  ));
  initialEstimate.insert(5, Pose2(21000, 21000, -M_PI_2));`

2. THE CODE THAT CRASH :
`  noiseModel::Diagonal::shared_ptr priorNoise = noiseModel::Diagonal::Sigmas(Vector3(3e-6, 3e-6, 1e-3));
  graph.emplace_shared<PriorFactor<Pose2> >(1, Pose2(0, 0, 0), priorNoise);

  // For simplicity, we will use the same noise model for odometry and loop closures
  noiseModel::Diagonal::shared_ptr model = noiseModel::Diagonal::Sigmas(Vector3(2e-6, 2e-6, 1e-3));

  // 2b. Add odometry factors
  // Create odometry (Between) factors between consecutive poses
  graph.emplace_shared<BetweenFactor<Pose2> >(1, 2, Pose2(200000, 0, 0     ), model);
  graph.emplace_shared<BetweenFactor<Pose2> >(2, 3, Pose2(200000, 0, M_PI_2), model);
  graph.emplace_shared<BetweenFactor<Pose2> >(3, 4, Pose2(200000, 0, M_PI_2), model);
  graph.emplace_shared<BetweenFactor<Pose2> >(4, 5, Pose2(200000, 0, M_PI_2), model);

  // 2c. Add the loop closure constraint
  // This factor encodes the fact that we have returned to the same pose. In real systems,
  // these constraints may be identified in many ways, such as appearance-based techniques
  // with camera images. We will use another Between Factor to enforce this constraint:
  graph.emplace_shared<BetweenFactor<Pose2> >(5, 2, Pose2(200000, 0, M_PI_2), model);
  graph.print("\nFactor Graph:\n"); // print

  // 3. Create the data structure to hold the initialEstimate estimate to the solution
  // For illustrative purposes, these have been deliberately set to incorrect values
  Values initialEstimate;
  initialEstimate.insert(1, Pose2(500000,     0,  0.2   ));
  initialEstimate.insert(2, Pose2(230000,  10000, -0.2   ));
  initialEstimate.insert(3, Pose2(410000,  10000,  M_PI_2));
  initialEstimate.insert(4, Pose2(400000, 200000,  M_PI  ));
  initialEstimate.insert(5, Pose2(210000, 210000, -M_PI_2));`

## Expected behavior

In my opinion, the problem is that the graph use big values with very little noise and approximation are done to be able to use float for example. 

You may be able to confirm my guess and peraps suggest solutions ? 

Thank you very much,
Clément

ProfFan, 2019-09-23:
#121

ProfFan, 2019-09-24:
@ClementLeBihan Can you share the code you are using on GitHub so we can take a look?

ProfFan, 2019-12-31:
I experimented with your example and the error is thrown here:

```
Partial Cholesky on HessianFactor failed.
Frontal keys Position 0: 5, 3, 4
HessianFactor:
 keys: 5(3) 3(3) 4(3)
Augmented information matrix: [
 	 2.50000756e+11       112232004  2.24448892e+13  5.57810201e-05      -112232004     -5.05048978   4.5924255e-05        -2.5e+11      -4.975e+16  2.44087647e+14;
    	      112232004  3.91666453e+11   2.8333066e+16 -1.03204984e-05  1.08333547e+11    0.0022673132         2.5e+11   4.5924255e-05         2.5e+14 -1.62631996e+15;
    	 2.24448892e+13   2.8333066e+16  5.66656832e+21     -7.28099848   2.1666934e+16      10554.4422               0               0        -1000000 -2.75252168e+20;
    	 5.57810201e-05 -1.03204984e-05     -7.28099848         2.5e+11  1.11998431e-06        -2.5e+14  -1.5308085e-05         2.5e+11               0        -2.5e+14;
    	     -112232004  1.08333547e+11   2.1666934e+16  1.11998431e-06  3.91666453e+11       4.975e+16        -2.5e+11  -1.5308085e-05               0  6.29648856e+14;
    	    -5.05048978    0.0022673132      10554.4422        -2.5e+14       4.975e+16      9.9005e+21      -4.975e+16        -2.5e+14        -1000000           5e+19;
    	  4.5924255e-05         2.5e+11               0  -1.5308085e-05        -2.5e+11      -4.975e+16           5e+11               0         2.5e+14          -5e+14;
    	       -2.5e+11   4.5924255e-05               0         2.5e+11  -1.5308085e-05        -2.5e+14               0           5e+11       4.975e+16          -5e+14;
    	     -4.975e+16         2.5e+14        -1000000               0               0        -1000000         2.5e+14       4.975e+16      9.9005e+21          -5e+19;
    	 2.44087647e+14 -1.62631996e+15 -2.75252168e+20        -2.5e+14  6.29648856e+14           5e+19          -5e+14          -5e+14          -5e+19  4.92386983e+19
  ]
/Users/proffan/Projects/Development/VISION/GTSAM/tests/testBigNumbers.cpp:95: Failure: "Exception:
Indeterminant linear system detected while working near variable
5 (Symbol: 5).
```

ProfFan, 2019-12-31:
Note that the determinant of the information matrix is 1.3*10^135, which is definitely too much for any numerical computations to happen. The core reason is that you need to have reasonable figures for the noise models...

closed 2019-12-31 by ProfFan