borglab/gtsam #120: CMAKE ERROR at wrap/CMakeLists.txt:29 (target_link_libraries):The "debug" argument must be followed by a library.

yuktimatta12, 2019-09-20:
On trying to build version 4.0.0 found here: https://bitbucket.org/gtborg/gtsam/get/357e739127de.zip

 I face the following issue:
CMAKE ERROR at wrap/CMakeLists.txt:29 (target_link_libraries):The "debug" argument must be followed by a library.

Below is what I see on the terminal window:
 The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/home/guhao/Documents/gtsam]
Boost 1.43 found.
Found Boost components:
serialization;system;filesystem;thread;program_options;date_time;timer;chrono;regex
-- GTSAM_BOOST_LIBRARIES: optimized;debug
-- Found TBB: /usr/include (found version "4.4") found components: tbb tbbmalloc
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES)
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp
CMake Warning (dev) at wrap/CMakeLists.txt:29 (target_link_libraries):
Link library type specifier "optimized" is followed by specifier "debug"
instead of a library name. The first specifier will be ignored.
This warning is for project developers. Use -Wno-dev to suppress it.

CMake Error at wrap/CMakeLists.txt:29 (target_link_libraries):
The "debug" argument must be followed by a library.

-- Configuring incomplete, errors occurred!
See also "/home/guhao/Documents/gtsam/build/CMakeFiles/CMakeOutput.log".
See also "/home/guhao/Documents/gtsam/build/CMakeFiles/CMakeError.log".

Any idea what could be going on?
I did try turning the GTSAM_BUILD_WRAP flag to off in the CMakeLists.txt file but the above error merely persists.

ProfFan, 2019-09-23:
Please try the latest develop and check if there is a similar problem. Also, please including steps for reproduction.

yuktimatta12, 2019-09-23:
Hi thanks for the reply.
I already tried it with this version: https://github.com/borglab/gtsam.git

And I still face the same issue.
Sorry what do you mean by the steps for reproduction.

Thanks for your help

ProfFan, 2019-09-23:
All the command line inputs and outputs.

You are absolutely welcome :)

yuktimatta12, 2019-09-23:
Please see below the command line inputs and outputs:

adas@adas-Latitude-E5550:/usr/local/gtsam-develop$ cd build
adas@adas-Latitude-E5550:/usr/local/gtsam-develop/build$ sudo cmake ..
[sudo] password for adas: 
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/usr/local/gtsam-develop]
-- Found Boost 1.70.0 at /usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET COMPONENTS serialization;system;filesystem;thread;program_options;date_time;timer;chrono;regex
-- Found boost_headers 1.70.0 at /usr/local/lib/cmake/boost_headers-1.70.0
-- Found boost_serialization 1.70.0 at /usr/local/lib/cmake/boost_serialization-1.70.0
--   libboost_serialization.so.1.70.0
-- Adding boost_serialization dependencies: headers
-- Found boost_system 1.70.0 at /usr/local/lib/cmake/boost_system-1.70.0
--   libboost_system.so.1.70.0
-- Adding boost_system dependencies: headers
-- Found boost_filesystem 1.70.0 at /usr/local/lib/cmake/boost_filesystem-1.70.0
--   libboost_filesystem.so.1.70.0
-- Adding boost_filesystem dependencies: headers
-- Found boost_thread 1.70.0 at /usr/local/lib/cmake/boost_thread-1.70.0
--   libboost_thread.so.1.70.0
-- Adding boost_thread dependencies: headers
-- Found boost_program_options 1.70.0 at /usr/local/lib/cmake/boost_program_options-1.70.0
--   libboost_program_options.so.1.70.0
-- Adding boost_program_options dependencies: headers
-- Found boost_date_time 1.70.0 at /usr/local/lib/cmake/boost_date_time-1.70.0
--   libboost_date_time.so.1.70.0
-- Adding boost_date_time dependencies: headers
-- Found boost_timer 1.70.0 at /usr/local/lib/cmake/boost_timer-1.70.0
--   libboost_timer.so.1.70.0
-- Adding boost_timer dependencies: chrono;headers
-- Found boost_chrono 1.70.0 at /usr/local/lib/cmake/boost_chrono-1.70.0
--   libboost_chrono.so.1.70.0
-- Adding boost_chrono dependencies: headers
-- Found boost_regex 1.70.0 at /usr/local/lib/cmake/boost_regex-1.70.0
--   libboost_regex.so.1.70.0
-- Adding boost_regex dependencies: headers
-- Found Boost: /usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.43") found components:  serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM_BOOST_LIBRARIES: optimized;/usr/local/lib/libboost_serialization.so.1.70.0;optimized;/usr/local/lib/libboost_system.so.1.70.0;optimized;/usr/local/lib/libboost_filesystem.so.1.70.0;optimized;/usr/local/lib/libboost_thread.so.1.70.0;optimized;/usr/local/lib/libboost_date_time.so.1.70.0;optimized;/usr/local/lib/libboost_regex.so.1.70.0;debug;debug;debug;debug;debug;debug
-- Found TBB: /usr/include (found suitable version "4.4", minimum required is "4.4") found components:  tbb tbbmalloc 
-- Found MKL: /opt/intel/mkl/include  
-- Found OpenMP_C: -fopenmp (found version "4.0") 
-- Found OpenMP_CXX: -fopenmp (found version "4.0") 
-- Found OpenMP: TRUE (found version "4.0")  
-- Found Eigen version: 3.3.7
CMake Error at wrap/CMakeLists.txt:32 (target_link_libraries):
  The "debug" argument must be followed by a library.

-- Configuring incomplete, errors occurred!
See also "/usr/local/gtsam-develop/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/gtsam-develop/build/CMakeFiles/CMakeError.log".

I don't know if this helps but when I try to changing the GTSAM_BUILD_WRAP flag to OFF in the gtsam-develop/CMakeLists.txt file, the above errors shifts from wrap/CMakeLists.txt to gtsam/CMakeLists.txt (though the error remains the same.)

ProfFan, 2019-09-23:
1. Why `sudo cmake`? Does not make much sense. Please check your local development environment.
2. What is your system version? CMake Version?

yuktimatta12, 2019-09-23:
1) The following site mentions sudo cmake as one of the steps to build:
https://gtsam.org/build/
2) I'm using Ubuntu version 16.04.6 LTS
CMake version 3.15

ProfFan, 2019-09-23:
@yuktimatta12 Are you sure? I searched for `sudo` and it is not there.

yuktimatta12, 2019-09-23:
Yes you're right. However if just type "cmake .." without sudo, I get 'permission denied' errors (as you can see below)
I've currently stored the gtsam-develop code in /usr/local which is why I tried it with sudo

adas@adas-Latitude-E5550:/usr/local/gtsam-develop/build$ cmake ..
CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:173 (file):
  file failed to open for writing (No such file or directory):

    /usr/local/gtsam-develop/build/CMakeFiles/CMakeOutput.log
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error: Could not open file for write in copy operation /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CMakeSystem.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:185 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:111 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:111 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:111 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:111 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:111 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:32 (CMAKE_DETERMINE_COMPILER_ID_BUILD)
  /usr/local/share/cmake-3.15/Modules/CMakeDetermineCXXCompiler.cmake:111 (CMAKE_DETERMINE_COMPILER_ID)
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeDetermineCompilerId.cmake:228 (file):
  file problem creating directory:
  /usr/local/gtsam-develop/build/CMakeFiles/3.15.2/CompilerIdCXX
Call Stack (most recent call first):
[... 188 lines cut ...]
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

CMake Error at /usr/local/share/cmake-3.15/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/c++"

  is not able to compile a simple test program.

  It fails with the following output:

    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:1 (project)

-- Configuring incomplete, errors occurred!
CMake Error: Cannot open file for write: /usr/local/gtsam-develop/build/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /usr/local/gtsam-develop/build/CMakeCache.txt
CMake Error: : System Error: Permission denied

closed 2019-09-23 by ProfFan

yuktimatta12, 2019-09-23:
Do you have any idea what the cmake error might be due to if not GTSAM?

ProfFan, 2019-09-23:
Privilege problems. Please refer to general documentations.