borglab/gtsam #121: Tracking Issue for 4.1 Release

ProfFan, 2019-09-23:
We are planning a 4.1 release, and this is the tracking issue to gather insights and feedbacks on that.

jlblancoc, 2019-09-23:
I think that using "milestones" is a much clearer way to keep track of pending issues before a new release... And there is already one for 4.1, so, should this one be closed?

ProfFan, 2019-09-23:
Milestones does not have a comment function. Also this method has been adopted by other large projects as well.

varunagrawal, 2019-09-24:
Instead of commenting on every issue to track, we should instead link to issues here, as well as tag each of the linked issues with the GTSAM 4.1 milestone.

ProfFan, 2019-09-24:
This model is borrowed from Rust, where you just use comment commands to manage issues and Tracking issues. Difference is that they have a bot that automates this process and we currently do not.

dellaert, 2019-09-24:
@ProfFan  is in charge of 4.1 and I defer to him as to what model to use...

varunagrawal, 2019-09-24:
But this affects how we do things going forward. If everyone uses a different model for release cycles when they have control, we're going to end up with code smell like the kind we saw over the summer (students in a rush to graduate, etc).

jlblancoc, 2019-09-24:
`<offtopic>`
> students in a rush to graduate

Ohhh, that sounds sooo familiar... It's sad to see that it's a universal thing in Academia
`</offtopic>`

ProfFan, 2019-09-24:
First sorry to everyone for being a bit aggressive on making changes here.

I agree with @varunagrawal  on the release cycle part. However this has nothing to do with release cycles :) Similar situations happen all over the open source software community, which is exactly the the thing that I am trying to prevent.

In larger organizations, a Request-For-Comments (RFC) based flow has been established for quite a long time, where people draft RFCs that contains all the rationale for a new function or breaking change, and all devs vote for merge or not merge. We are smaller so IDK if that is too much hassle.

dellaert, 2020-07-22:
Ok, now 4.0.3 is pretty much ready? We should clean up this issue: IMHO it should just be the pybind wrapper in 4.1 as well as Removing deprecated methods/functions.

closed 2021-07-19 by ProfFan