borglab/gtsam #122: Tracking use-after-frees with Clang's ASAN

ProfFan, 2019-09-24:
This is a mega-thread used for resolving bugs found by the LLVM sanitizer.

ProfFan, 2019-09-24:
https://gist.github.com/ProfFan/0a753ea15de807dd58f28b7518fc1eb6

ProfFan, 2019-09-24:
Note that we should have two things for improving stability:
- A fuzzer (afl or so) and a build machine that fuzz the library day after day
- A mechanism for detection of hidden memory leaks, for example, make the `TEST` macros run in a loop that does the test 1000 times, and observe the memory footprint. Current test suite runs in ~20s, so 1000*20=20000s=6h is pretty reasonable.

dellaert, 2019-09-24:
Yowza. Let's talk tomorrow.

ProfFan, 2019-09-24:
@dellaert Roger that

ProfFan, 2019-09-24:
Aiming to have AddressSanitizer and LeakSanitizer in Travis and in Tests.

#121