borglab/gtsam #123: Can't compile gtsam_develop examples  in Visual Studio 2012

HesperZhao, 2019-09-24:
Hi, I am using CMake 3.14.0, boost1.69.0 and Visual Studio 2012 express. In the process, when I open the project that generated in the CMake I can't build every .lib and .cpp file. For example, I can successfully compile CppUniteLite.lb and metis.lib. When I build the other .lib files and examples like Pose2SLAMExample.exe I got a bunch of grammatical errors.

Could someone tell me what can I do?

Any help you can provide is greatly appreciated.

ProfFan, 2019-09-24:
I do not think Visual Studio 2012 has proper C++11 support.

jlblancoc, 2019-09-24:
@ProfFan is right, take a look [here](https://en.cppreference.com/w/cpp/compiler_support). 
I would say that MSVC 2015 *may* be the minimum MSVC supported version for GTSAM, but MSVC tends to be so buggy that it might even be a more recent version.

dellaert, 2019-09-25:
I think we should make it clear VS 2017 is required.

HesperZhao, 2019-09-25:
Thanks!!! 
I am gonna try at once.

HesperZhao, 2019-09-25:
I use Visual Studio 2017 and it works!!!!

But I still have some questions which I don't know matters or not. When I use CMake to generate the GTSAM project, it reminds me could not find TBB, MKL, DOXYGN, and GeographicLib. Do I need to set some Entry in CMake or anything else? In addition to this, I get a warning which is ignoring Boost restriction on optional lvalue assignment from rvalue. Besides, I don't understand why CMake appears disabled to install MATLAB toolbox. I already successfully debug the program, so I'm even more confused as to why these problems occur and how I can solve them.

Thank you again for your help.

ProfFan, 2019-09-25:
Windows really does not have a proper library search path system. You need to locate the corresponding libraries and supply them to CMake yourself. Also `boost` does not really like MSVC, but there is a "compliant" mode in VS that you can try, just google it.

> Besides, I don't understand why CMake appears disabled to install MATLAB toolbox.

No idea what you are talking about here.

closed 2019-09-25 by ProfFan