borglab/gtsam #124: CMake Compiler Identification Strategy Change

ProfFan, 2019-09-25:
## Description

`CMP0025`
Compiler id for Apple Clang is now AppleClang.

CMake 3.0 and above recognize that Apple Clang is a different compiler than upstream Clang and that they have different version numbers. CMake now prefers to present this to projects by setting the `CMAKE_<LANG>_COMPILER_ID` variable to AppleClang instead of Clang. However, existing projects may assume the compiler id for Apple Clang is just Clang as it was in CMake versions prior to 3.0. Therefore this policy determines for Apple Clang which compiler id to report in the `CMAKE_<LANG>_COMPILER_ID` variable after language `<LANG>` is enabled by the project() or enable_language() command. The policy must be set prior to the invocation of either command.

The OLD behavior for this policy is to use compiler id Clang. The NEW behavior for this policy is to use compiler id AppleClang.

This policy was introduced in CMake version 3.0. Use the cmake_policy() command to set this policy to OLD or NEW explicitly. Unlike most policies, CMake version 3.11.4 does not warn by default when this policy is not set and simply uses OLD behavior. See documentation of the CMAKE_POLICY_WARNING_CMP0025 variable to control the warning.

## Steps to reproduce

1. Set `PATH` to prioritize Clang instead of Apple Clang
2. `cmake ../GTSAM -DGTSAM_BUILD_UNSTABLE=OFF -GNinja -DBOOST_ROOT="/usr/local/Cellar/boost@1.68/1.68.0" -DTBB_ROOT="/usr/local/Cellar/tbb/2019_U8" -DCMAKE_BUILD_TYPE=Debug -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang`

Result:

```
CMake Error at cmake/GtsamBuildTypes.cmake:228 (target_compile_features):
  target_compile_features no known features for CXX compiler

  "Clang"

  version 8.0.1.
Call Stack (most recent call first):
  cmake/GtsamTesting.cmake:290 (gtsam_apply_build_flags)
  cmake/GtsamTesting.cmake:80 (gtsamAddExesGlob_impl)
  timing/CMakeLists.txt:1 (gtsamAddTimingGlob)
```

## Expected behavior

Configuration succeeds.

## Environment

Clang 8.0.1 on macOS 10.14

## Additional information

PR underway.

ProfFan, 2019-09-25:
Apple's Clang is lagging the LLVM version by about 1-2 years, so many advanced features does not exist.

jlblancoc, 2019-10-13:
@ProfFan : According to [this](https://gitlab.kitware.com/cmake/cmake/issues/19250), adding: 

```cmake
if(POLICY CMP0025)
  cmake_policy(SET CMP0025 NEW)
endif()
```

Should fix it. I don't have OSX for testing, could you give it a try?
That code could be inserted at the beginning of `GtsamBuildTypes.cmake`, for example.

closed 2020-07-12 by varunagrawal