borglab/gtsam #125: Expression support in MATLAB/Python

varunagrawal, 2019-09-27:
## Feature

Currently, the wrappers don't have support for Expressions. It would be great to have this supported so that one can work with these features at a higher level.

## Motivation

Expressions are pretty awesome?

## Pitch

## Alternatives

No alternative exists.

## Additional context

ProfFan, 2019-09-29:
Could be a little bit difficult with the current python I/F. I think to implement this we need to associate `Expression` objects in C++ and Python. Also if we want new non-arithmetic ops we need to either implement in C++ or Cythonize a function and pass ptr to C++.

varunagrawal, 2020-12-01:
This is not really a must-have feature so I am closing this.

closed 2020-12-01 by varunagrawal