borglab/gtsam #126: Compilation of MATLAB toolbox fails for non-default MATLAB paths (Unix only)

VGab, 2019-09-30:
While compiling GTSAM and the dedicated matlab toolbox a few hours ago, I encountered the issue that ``mex.h`` could not have been found. 

While this bug stems from a customized problem at my PC, the quick bug fix may be of help for others or can be added as an additional options to the CMake files in a future release

## Description

I followed the instructions on the installation page as well as the dedicated readme in the matlab sections. Nonethless, the compilation terminated as seen below
[image]

## Steps to reproduce

The problem in here lies presumably in my custom setup, where MATLAB is not installed in the standard unix location, i.e. ``/usr/local/MatlabRelease/``, but mounted from a network share within a lab environment.
So in order to reproduce this error,

1.  alter the location of Matlab during installation to 
    ``` sh
     mv /usr/local/MatlabR20DDs /mnt/MatlabR20DDs
    ```
2. follow the steps from the installation section and run the compilation process.

## Environment

The problem occurred on

- Linux Kubuntu 18.04 LTS system (Kernel 4.15.0-55-generic
- Python interpreter set to Python 2.7.15+ 
- Matlab version R2019a
- gcc-version (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
- cmake version 3.10.2

## Additional information
### quick and dirty fix

The compilation was successfully run after adding the Matlab c-headers directly into the CMakeLists.txt file in line 401 and 424

```
# Build wrap
if (GTSAM_BUILD_WRAP)
    include_directories("/mnt/MatlabR20DDs/extern/include")
    add_subdirectory(wrap)
endif(GTSAM_BUILD_WRAP)
[...]
# Matlab toolbox
if (GTSAM_INSTALL_MATLAB_TOOLBOX)
    include_directories("/mnt/MatlabR20DDs/extern/include")
    add_subdirectory(matlab)
endif()
```

Preferably one could introduce a new option to the CMakeLists, but I assumed this to be a very specific and rare error, thus preferring to pinpoint to the quick fix above.

VGab, 2019-09-30:
Sorry in case this bug is better to be be posted in the google group instead. But as I did not have access over there i was afraid of forgetting to report the problem otherwise.

varunagrawal, 2019-09-30:
Or you could just update the CMake Prefix Path before running `cmake`.

```
export CMAKE_PREFIX_PATH=/mnt/MatlabR20DDs/extern:$CMAKE_PREFIX_PATH
```

https://gitlab.kitware.com/cmake/community/wikis/doc/cmake/Useful-Variables

varunagrawal, 2019-09-30:
This is pretty much a custom case for you and can be easily fixed with some cmake cleverness, rather than a GTSAM bug. I vote to close this.

dellaert, 2019-10-01:
Thanks @VGab and @varunagrawal ! I think we should not close it yet. At the very least we should see whether Varun’s suggestion works, and then we should add it into docs and on the website. but, actually, i’m not in love with  doing this via `export`. Can this be done on the cmake command line as well?

mikesheffler, 2019-10-01:
Looking at `CMake-gui` on Windows, I see `MATLAB_ROOT` set to `C:/Program Files/MATLAB/R2019a`.

So, would

`$ cmake -DMATLAB_ROOT=:/Program Files/MATLAB/R2019a`

be the shell equivalent (let's pretend this is Linux for a moment)? I'm awful at following CMake rabbit holes ...

VGab, 2019-10-01:
Thanks for denoting this @mikesheffler. I was not aware of that as the MATLAB ROOT variable is not shown by default in ``cmake`` but only after an initial conifguration with the ``GTSAM_INSTALL_MATLAB_TOOLBOX`` flag set to `ON``.
Nonetheless, running

```sh
cmake -DMATLAB_ROOT=/mnt/MatlabR20DDs  -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON  ..
```

circumvents the nonsense I claimed above and does resolve this "issue". I support @varunagrawal's claim that this issue is more of a rare and custom case problem and can be closed, but in order to make this problem visible to all users, one could add 

```
if (${GTSAM_INSTALL_MATLAB_TOOLBOX})
	message("MATLAB ROOT set to ${MATLAB_ROOT}")
endif()
```

in line 579-581 at the top-level CMakeLists.txt. I think this is not worth a pull request and leave it up to you if this should be added or not.

ProfFan, 2019-10-01:
Think we should print MEX compiler and MATLAB path as @VGab suggested, as this is also what OpenCV does. I will prepare a PR shortly.

varunagrawal, 2020-07-10:
The CMake printing was added by #393.

closed 2020-07-10 by varunagrawal