borglab/gtsam #138: timeGaussianFactorGraph.cpp fails to compile

ptrmu, 2019-10-09:
## Description

When building the target timeGaussianFactorGraph on branch develop commit 2f6edeeb416a812719e5f78520791a7368d588ce I received the following error:

.../gtsam-repo/timing/timeGaussianFactorGraph.cpp:43:43: error: ‘struct std::pair<gtsam::GaussianFactorGraph, gtsam::VectorValues>’ has no member named ‘get’
   GaussianFactorGraph fg = planarGraph(N).get<0>();

changing "get<0>()" to "first" fixed the compile error.

ProfFan, 2019-10-09:
Strange, please kindly provide you environment (OS Version, GCC Version, boost Version, etc.) and your command line building GTSAM with all outputs.

ptrmu, 2019-10-09:
Ubuntu 18.04
gcc 7.4.0
boost 1.65.1

using cmake-gui this is the configuration:
===============================================================
================  Configuration Options  ======================
  CMAKE_CXX_COMPILER_ID type     : GNU
  CMAKE_CXX_COMPILER_VERSION     : 7.4.0
  CMake version                  : 3.10.2
  CMake generator                : Unix Makefiles
  CMake build tool               : /usr/bin/make
Build flags                                               
  Build Tests                    : Enabled
  Build examples with 'make all' : Enabled
  Build timing scripts with 'make all': Enabled
  Build shared GTSAM libraries   : Enabled
  Put build type in library name : Enabled
  Build libgtsam_unstable        : Enabled
  Build for native architecture  : Enabled
  Build type                     : Debug
  C compilation flags            :  -g
  C++ compilation flags          :  -g
 GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_11
 GTSAM_COMPILE_OPTIONS_PRIVATE                    : -Wall;$<$<CONFIG:Debug>:-g;-fno-inline>;$<$<CONFIG:Release>:-O3>;$<$<CONFIG:Timing>:-g;-O3>;$<$<CONFIG:Profiling>:-O3>;$<$<CONFIG:RelWithDebInfo>:-g;-O3>;-Wno-unused-local-typedefs
 GTSAM_COMPILE_OPTIONS_PUBLIC                     : -march=native
 GTSAM_COMPILE_DEFINITIONS_PRIVATE                : $<$<CONFIG:Debug>:_DEBUG;EIGEN_INITIALIZE_MATRICES_BY_NAN>;$<$<CONFIG:Release>:NDEBUG>;$<$<CONFIG:Timing>:NDEBUG;ENABLE_TIMING>;$<$<CONFIG:Profiling>:NDEBUG>;$<$<CONFIG:RelWithDebInfo>:NDEBUG>
 GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : BOOST_OPTIONAL_ALLOW_BINDING_TO_RVALUES;BOOST_OPTIONAL_CONFIG_ALLOW_BINDING_TO_RVALUES
 GTSAM_COMPILE_OPTIONS_PRIVATE_DEBUG              : -g;-fno-inline
 GTSAM_COMPILE_OPTIONS_PUBLIC_DEBUG               : 
 GTSAM_COMPILE_DEFINITIONS_PRIVATE_DEBUG          : _DEBUG;EIGEN_INITIALIZE_MATRICES_BY_NAN
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_DEBUG           : 
 GTSAM_COMPILE_OPTIONS_PRIVATE_RELEASE            : -O3
 GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
 GTSAM_COMPILE_DEFINITIONS_PRIVATE_RELEASE        : NDEBUG
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
 GTSAM_COMPILE_OPTIONS_PRIVATE_TIMING             : -g;-O3
 GTSAM_COMPILE_OPTIONS_PUBLIC_TIMING              : 
 GTSAM_COMPILE_DEFINITIONS_PRIVATE_TIMING         : NDEBUG;ENABLE_TIMING
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_TIMING          : 
 GTSAM_COMPILE_OPTIONS_PRIVATE_PROFILING          : -O3
 GTSAM_COMPILE_OPTIONS_PUBLIC_PROFILING           : 
 GTSAM_COMPILE_DEFINITIONS_PRIVATE_PROFILING      : NDEBUG
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_PROFILING       : 
 GTSAM_COMPILE_OPTIONS_PRIVATE_RELWITHDEBINFO     : -g;-O3
 GTSAM_COMPILE_OPTIONS_PUBLIC_RELWITHDEBINFO      : 
 GTSAM_COMPILE_DEFINITIONS_PRIVATE_RELWITHDEBINFO : NDEBUG
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELWITHDEBINFO  : 
 GTSAM_COMPILE_OPTIONS_PRIVATE_MINSIZEREL         : 
 GTSAM_COMPILE_OPTIONS_PUBLIC_MINSIZEREL          : 
 GTSAM_COMPILE_DEFINITIONS_PRIVATE_MINSIZEREL     : 
 GTSAM_COMPILE_DEFINITIONS_PUBLIC_MINSIZEREL      : 
  Use System Eigen               : OFF (Using version: 3.3.7)
  Use Intel TBB                  : TBB not found
  Eigen will use MKL             : MKL not found
  Eigen will use MKL and OpenMP  : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
  Default allocator              : STL
  Build with ccache              : No
Packaging flags                                               
  CPack Source Generator         : TGZ
  CPack Generator                : TGZ
GTSAM flags                                               
  Quaternions as default Rot3     : Disabled
  Runtime consistency checking    : Disabled
  Rot3 retract is full ExpMap     : Enabled
  Pose3 retract is full ExpMap    : Disabled
  Deprecated in GTSAM 4 allowed   : Disabled
  Point3 is typedef to Vector3    : Disabled
  Metis-based Nested Dissection   : Enabled
  Use tangent-space preintegration: Enabled
  Build Wrap                     : Enabled
MATLAB toolbox flags                                      
  Install matlab toolbox         : Disabled
Cython toolbox flags                                      
  Install Cython toolbox         : Disabled
===============================================================

In the file timeGaussianFactorGraph.cpp the function planarGraph(N) returns a std::pair<> class. This class does not define the get() member that is being referenced. Maybe at some point planarGraph(N) returned a tuple. But this file was not updated when planarGraph(N) was changed to return a std::pair().

Thanks!

dellaert, 2019-10-11:
@ptrmu I can see that, could you just create a quick PR?

dellaert, 2019-10-13:
Thanks for pr!!

closed 2019-10-13 by dellaert