borglab/gtsam #141: Expose GPSFactor and GPSFactor2 in the Python wrapper

octopuscabbage, 2019-10-10:
## Feature

We would like GPSFactor and GPSFactor2 exposed in the python API. 
Currently we have tried wrapping it ourselves in gtsam.h but are having problems with building it.

## Motivation

It would be useful to be able to use the full NavState in the python API. Currently only the IMU factors are implemented.

## Pitch

GPSFactor and GPSFactor2 to be usable in the python wrapper.

## Alternatives

Expose the ability to create custom factors in the python API.

## Additional context

None

dellaert, 2019-10-11:
I would love to assist. Could you create a PR with your attempt at wrapping?
By the way, the language that gtsam.h accepts is close to c++ but not quite. See comments at top. My approach to debugging it is to start with an empty class and then add methods one by one to find problem. Error reporting is terrible, but we are working on a new wrap...

octopuscabbage, 2019-10-11:
Will do, I'll talk with my colleague and put what we have so far in a fork. 

We read the preamble and think our code conforms to it but yeah it has some weird bad parsing error.

octopuscabbage, 2019-10-11:
We actually got it to build - we'll post our PR later.

dellaert, 2019-10-11:
Excellent!! Look forward to it 👍

dellaert, 2019-10-19:
@octopuscabbage don't forget the PR :-)

dellaert, 2019-10-21:
Fixed by PR #155 . Thanks @MandyXie !

closed 2019-10-21 by dellaert