borglab/gtsam #142: IndeterminantLinearSystemException

RobotGithubLee, 2019-10-11:
Indeterminant linear system detected while working near variable
8646911284551352615 (Symbol: x295).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.

varunagrawal, 2019-10-11:
Ummm what's the actual issue here? Seems more like an implementation problem on your end rather than a big in GTSAM.

You might want to instead ask on the Google group.

dellaert, 2019-10-11:
I agree. Please try the Google group, but even there more information will be needed, and what debugging steps you took...

closed 2019-10-11 by dellaert