borglab/gtsam #145: CMake Build Failure ["debug" argument must be followed by a library]

acxz, 2019-10-12:
## Description

gtsam does not build, in fact `cmake` fails.

## Steps to reproduce

1. Clone gtsam
2. `cd gtsam && mkdir build && cd build`
3. `cmake ..`

Error Message:
```
-- The CXX compiler identification is GNU 9.2.0
-- The C compiler identification is GNU 9.2.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_SOURCE_ROOT_DIR: [/home/acxz/vcs/github/borglab/og/gtsam]
-- Found Boost: /usr/lib64/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.43") found components:  serialization system filesystem thread program_options date_time timer chrono regex 
-- GTSAM_BOOST_LIBRARIES: optimized;/usr/lib64/libboost_serialization.so.1.71.0;optimized;/usr/lib64/libboost_system.so.1.71.0;optimized;/usr/lib64/libboost_filesystem.so.1.71.0;optimized;/usr/lib64/libboost_thread.so.1.71.0;optimized;/usr/lib64/libboost_date_time.so.1.71.0;optimized;/usr/lib64/libboost_regex.so.1.71.0;debug;debug;debug;debug;debug;debug
-- Found TBB: /usr/include (found suitable version "2019.0", minimum required is "4.4") found components:  tbb tbbmalloc 
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found Eigen version: 3.3.7
CMake Error at wrap/CMakeLists.txt:32 (target_link_libraries):
  The "debug" argument must be followed by a library.

-- Configuring incomplete, errors occurred!
```

## Expected behavior

`cmake` command should not error out and finish successfully.

## Environment

gtsam: `develop` branch
cmake: `3.15.4`
boost: `1.71.0`
OS: ArchLinux

## Additional information

After some prelim debugging, it seems as tho in the latest `cmake` version (or the latest `boost` version), you cannot have empty fields after library identifies such as `debug`. Since I am not building gtsam in debug mode, the following [variables](https://github.com/borglab/gtsam/blob/develop/wrap/CMakeLists.txt#L9-L11) are empty for me and as such when calling `${WRAP_BOOST_LIBRARIES}` further in the CMakeLists.txt, the above error triggers. This happens not just in the `wrap/CMakeLists.txt` but also in other locations where `debug` is specified in the variable name, but no debug libraries are included. Another case is with `${GTSAM_BOOST_LIBRARIES}` [here](https://github.com/borglab/gtsam/blob/develop/gtsam/CMakeLists.txt#L108) with the relevant definition happening [here](https://github.com/borglab/gtsam/blob/develop/CMakeLists.txt#L181-L194).

jlblancoc, 2019-10-12:
I think PR https://github.com/borglab/gtsam/pull/135 should fix this one too. 
Please, give a try to that PR and confirm it to us.

acxz, 2019-10-12:
Just checked the PR and cmake works fine! Hopefully this PR can get merged ASAP and a new release made, so ArchLinux users can have a working gtsam package.

dellaert, 2019-10-12:
I merged #135 and will close this issue for you :-) Thanks for helping to make GTSAM better!

closed 2019-10-12 by dellaert

acxz, 2019-10-12:
Thanks for the super quick response, really means a lot!

shaoxh, 2021-10-19:
Hi, I was using tag 4.0.0 but this problem is still there.
Error message like ...
CMake Warning (dev) at wrap/CMakeLists.txt:29 (target_link_libraries):
  Link library type specifier "optimized" is followed by specifier "debug"
  instead of a library name.  The first specifier will be ignored.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at wrap/CMakeLists.txt:29 (target_link_libraries):
  The "debug" argument must be followed by a library.

Could you tell me which tag has fixed your error? @acxz