borglab/gtsam #148: Build fails: error: static_assert failed due to requirement 'boost::is_lvalue_reference<const gtsam::Scatter &&>::value' "binding rvalue references to optional lvalue references is disallowed"

yurivict, 2019-10-13:
```
In file included from /usr/local/include/boost/optional/optional.hpp:1468:
/usr/local/include/boost/optional/detail/optional_reference_spec.hpp:32:5: error: static_assert failed due to requirement 'boost::is_lvalue_reference<const gtsam::Scatter &&>::value' "binding rvalue references to optional lvalue references is disallowed"
    BOOST_STATIC_ASSERT_MSG(boost::is_lvalue_reference<From>::value, 
    ^                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/boost/static_assert.hpp:31:45: note: expanded from macro 'BOOST_STATIC_ASSERT_MSG'
#     define BOOST_STATIC_ASSERT_MSG( ... ) static_assert(__VA_ARGS__)
                                            ^             ~~~~~~~~~~~
/usr/local/include/boost/optional/detail/optional_reference_spec.hpp:184:54: note: in instantiation of function template specialization 'boost::detail::prevent_binding_rvalue<const gtsam::Scatter &&>' requested here
    optional(T&& /* rhs */) BOOST_NOEXCEPT { detail::prevent_binding_rvalue<T&&>(); }
                                                     ^
/usr/local/include/boost/smart_ptr/make_shared_object.hpp:256:20: note: in instantiation of member function 'boost::optional<const gtsam::Scatter &>::optional' requested here
    ::new( pv ) T( boost::detail::sp_forward<Args>( args )... );
                   ^
/usr/ports/misc/gtsam/work/gtsam-3.2.3/gtsam/linear/HessianFactor.cpp:628:26: note: in instantiation of function template specialization 'boost::make_shared<gtsam::HessianFactor, const gtsam::GaussianFactorGraph &, gtsam::Scatter>' requested here
    jointFactor = boost::make_shared<HessianFactor>(factors, Scatter(factors, keys));
                         ^
1 error generated.
[153/262] /usr/bin/c++  -DBOOST_OPTIONAL_ALLOW_BINDING_TO_RVALUES -Dgtsam_EXPORTS -I/usr/ports/misc/gtsam/work/gtsam-3.2.3/CppUnitLite -I. -I/usr/ports/misc/gtsam/work/gtsam-3.2.3 -I/usr/ports/misc/gtsam/work/gtsam-3.2.3/gtsam/3rdparty/CCOLAMD/Include -I/usr/ports/misc/gtsam/work/gtsam-3.2.3/gtsam/3rdparty/UFconfig -I/usr/local/include -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer -ftemplate-depth=1024 -O3 -Wall -DNDEBUG -Wall -fPIC -MD -MT gtsam/CMakeFiles/gtsam.dir/linear/IterativeSolver.cpp.o -MF gtsam/CMakeFiles/gtsam.dir/linear/IterativeSolver.cpp.o.d -o gtsam/CMakeFiles/gtsam.dir/linear/IterativeSolver.cpp.o -c /usr/ports/misc/gtsam/work/gtsam-3.2.3/gtsam/linear/IterativeSolver.cpp
```

FreeBSD 12
clang 8

jlblancoc, 2019-10-13:
What's the version of Boost you installed under `/usr/local`?

dellaert, 2019-10-13:
@gchenfc your PR would fix this.

yurivict, 2019-10-13:
Boost is of version ```1.71.0```.

yurivict, 2019-10-13:
I tried to create a FreeBSD port for ```gtsam``` but got stuck because of this issue.

dellaert, 2019-10-19:
@gchenfc ?

PR #156 (2019-11-13): boost::optional<const Ordering &>
Get rid of boost::optional, this PR is for Ordering and Scatter classes (should fix #148 ).
Replace all usages of boost::optional<\*Ordering\*> with function overloads.

compiles and passes tests, but some potentially code-breaking changes in:

- Marginals.h - order of arguments had to change since `factorization` has a default value
- EliminatableFactorGraph.h - marginalMultifrontalBayesNet and marginalMultifrontalBayesTree no longer accept `boost::none` as a placeholder to specify later arguments
- NonlinearFactorGraph.h - same deal

Notes:

- EliminateableF [... cut]
closed 2019-11-13 by ProfFan