borglab/gtsam #160: Marginals computation is erroneous for 3D poses

YB27, 2019-11-06:
## Description
Marginals computation gives wrong covariances when using factor graph in 3D case with no loop closure.

## Steps to reproduce
To test this, I have rewritten the OdometryExample.cpp using 3D poses :

```cpp
  using namespace std;
  using namespace gtsam;

  // Create an empty nonlinear factor graph
  NonlinearFactorGraph graph;

  // Add a prior on the first pose, setting it to the origin
  // A prior factor consists of a mean and a noise model (covariance matrix)
  Rot3 identity = Rot3::Ypr(0.0, 0.0, 0.0);
  Pose3 priorMean(identity, Point3::identity()); // prior at origin
  Vector6 priorSigmas;
  priorSigmas << 0.3, 0.3, 0.001, 0.1, 0.001, 0.001;
  noiseModel::Diagonal::shared_ptr priorNoise = noiseModel::Diagonal::Sigmas(priorSigmas);
  graph.emplace_shared<PriorFactor<Pose3> >(1, priorMean, priorNoise);

  // Add odometry factors
  Point3 odometry_t(2.0, 0.0, 0.0);
  Pose3 odometry(identity, odometry_t);
  // For simplicity, we will use the same noise model for each odometry factor
  Vector6 odometrySigmas;
  odometrySigmas << 0.2, 0.2, 0.001, 0.1, 0.001, 0.001;
  noiseModel::Diagonal::shared_ptr odometryNoise =  noiseModel::Diagonal::Sigmas(odometrySigmas);
  // Create odometry (Between) factors between consecutive poses
  graph.emplace_shared<BetweenFactor<Pose3> >(1, 2, odometry, odometryNoise);
  graph.emplace_shared<BetweenFactor<Pose3> >(2, 3, odometry, odometryNoise);
  graph.print("\nFactor Graph:\n"); // print

  // Create the data structure to hold the initialEstimate estimate to the solution
  // For illustrative purposes, these have been deliberately set to incorrect values
  Values initial;
  initial.insert(1, Pose3(Rot3::Ypr(0.2  , 0.1 , -0.2) , Point3(0.5, 0.0, 0.1)));
  initial.insert(2, Pose3(Rot3::Ypr(-0.2 , -0.1, -0.1) , Point3(2.3, 0.1, 0.1)));
  initial.insert(3, Pose3(Rot3::Ypr(0.1  , 0.1 , 0.1 ) , Point3(4.1, 0.1, 0.1)));
  initial.print("\nInitial Estimate:\n"); // print

  // optimize using Levenberg-Marquardt optimization
  Values result = LevenbergMarquardtOptimizer(graph, initial).optimize();
  result.print("Final Result:\n");

  // Calculate and print marginal covariances for all variables
  cout.precision(2);
  Marginals marginals(graph, result);
  cout << "x1 covariance:\n" << marginals.marginalCovariance(1) << endl;
  cout << "x2 covariance:\n" << marginals.marginalCovariance(2) << endl;
  cout << "x3 covariance:\n" << marginals.marginalCovariance(3) << endl;

  return 0;
```
Note that the z, pitch and roll covariances are set to small values in order to compare with the results in the 2D case.

Running the above code outputs the following covariances  (Poses are correct) :
```console
x1 covariance:
    0.09 -5.1e-29 -9.3e-33  1.7e-46  3.4e-34   -3e-34
-5.1e-29     0.09 -2.3e-33  2.9e-61  1.2e-34  1.7e-61
-9.3e-33 -2.3e-33    1e-06 -1.3e-34 -3.7e-65  3.1e-65
 1.7e-46  2.9e-61 -1.3e-34     0.01 -6.5e-35 -5.5e-79
 3.4e-34  1.2e-34 -3.7e-65 -6.5e-35    1e-06 -1.1e-66
  -3e-34  1.7e-61  3.1e-65 -5.5e-79 -1.1e-66    1e-06
x2 covariance:
    0.13 -5.2e-18  1.2e-17 -2.1e-37 -6.2e-21 -1.1e-21
-5.2e-18     0.13 -1.7e-22  2.3e-17 -3.4e-22    -0.18
 1.2e-17 -1.7e-22    2e-06  1.2e-22    2e-06  3.4e-22
-2.1e-37  2.3e-17  1.2e-22     0.02 -5.8e-19 -4.5e-17
-6.2e-21 -3.4e-22    2e-06 -5.8e-19    6e-06  6.4e-22
-1.1e-21    -0.18  3.4e-22 -4.5e-17  6.4e-22     0.36
x3 covariance:
    0.17 -1.6e-17  1.2e-17  1.4e-33  1.8e-17    1e-17
-1.6e-17     0.17 -3.2e-22  2.2e-17 -3.5e-18    -0.44
 1.2e-17 -3.2e-22    3e-06  4.6e-22    6e-06  1.2e-21
 1.4e-33  2.2e-17  4.6e-22     0.03 -2.3e-18 -8.4e-17
 1.8e-17 -3.5e-18    6e-06 -2.3e-18  2.3e-05  1.4e-17
   1e-17    -0.44  1.2e-21 -8.4e-17  1.4e-17      1.6
```
## Expected behavior
I expected a result similar to the 2D case (in OdometryExample.cpp) : 
```console
x2 covariance:
     0.13   4.7e-18  2.4e-18
4.7e-18        0.17       0.02     
2.4e-18        0.02       0.02     
x3 covariance:
     0.17 2.7e-17  8.4e-18  
2.7e-17      0.37       0.06  
8.4e-18      0.06       0.03  
```  
For x2 and x3 in the 3D case, the first and second diagonal elements are equal (resp. 0.13 and 0.17) whereas the second element (covariance on y) should be higher (as in the 2D above).

Furthermore, the roll covariance (last diagonal element) is also growing very large.

## Environment
Ubuntu 18.04
gtsam 4.0.2
C++ gcc 7.4 using system Eigen 3.3.7 and system Boost 1.67.0
TBB and MKL disabled

## Additional information

michaelbosse, 2019-11-11:
I believe that GTSAM uses the convention of having rotation then translation in the order for its tangent spaces.  Thus the components for setting the `priorSigmas` and `odometrySigmas` would need to be permuted.

ptrmu, 2019-12-04:
Using the modified lines:
###
    priorSigmas << 0.001, 0.001, 0.1, 0.3, 0.3, 0.0001;
    odometrySigmas << 0.001, 0.001, 0.1, 0.2, 0.2, 0.0001;
###

does produce the same covariances as the 2D case. Even the non-zero correlation between yaw and y.

YB27, 2019-12-05:
Thanks for your response. I tried the permutation and I effectively obtain the expected values.
So it means that for sigma, the order is [roll pitch yaw x y z] ? It may be interesting to notice this convention somewhere in the documentation. 
By the way, it is inconsistent with the 2D order [x y yaw].

mikesheffler, 2019-12-05:
> So it means that for sigma, the order is `[roll pitch yaw x y z]` ? It may be interesting to notice this convention somewhere in the documentation.

It's there, it's just not obvious. Try looking at places where homogeneous matrices are constructed from individual values or vice versa. For example, the definition of the static method `Expmap` in `Pose2` and `Pose3`.

There are also purpose-built helper function that can help you find what you need. In [Pose3.h](https://github.com/borglab/gtsam/blob/develop/gtsam/geometry/Pose3.h#L304), there is
```
  /**
   * Return the start and end indices (inclusive) of the translation component of the
   * exponential map parameterization
   * @return a pair of [start, end] indices into the tangent space vector
   */
  inline static std::pair<size_t, size_t> translationInterval() {
    return std::make_pair(3, 5);
  }

  /**
   * Return the start and end indices (inclusive) of the rotation component of the
   * exponential map parameterization
   * @return a pair of [start, end] indices into the tangent space vector
   */
  static std::pair<size_t, size_t> rotationInterval() {
    return std::make_pair(0, 2);
  }
```
and, in [Pose2.h](https://github.com/borglab/gtsam/blob/develop/gtsam/geometry/Pose2.h#L277), there is
```
  /**
   * Return the start and end indices (inclusive) of the translation component of the
   * exponential map parameterization
   * @return a pair of [start, end] indices into the tangent space vector
   */
  inline static std::pair<size_t, size_t> translationInterval() { return std::make_pair(0, 1); }

  /**
   * Return the start and end indices (inclusive) of the rotation component of the
   * exponential map parameterization
   * @return a pair of [start, end] indices into the tangent space vector
   */
  static std::pair<size_t, size_t> rotationInterval() { return std::make_pair(2, 2); }
```
> By the way, it is inconsistent with the 2D order [x y yaw].

Yeah, there is a historical reason for this, I think, and I don't remember what it might be.

YB27, 2019-12-05:
Thanks for your precision. My problem is now solved !
I didn't assume that Lie group and the Lie algebra would have the same convention order (it's logical but it is not a proof)
I think that explicitly noting the convention used in the class documentation would make things clearer.

closed 2019-12-05 by YB27

dellaert, 2019-12-05:
The "historical reason" for `x, y, theta` order is because all 2D SLAM papers use that convention :-)