borglab/gtsam #161: Default value for Point3

varunagrawal, 2019-11-08:
## Description

The `Point3` class when initialized with a default constructor such as `Point3()` is not allocated and defaults to `NaN`s which can cause issues in the final result without explicit code failure.

This is because Eigen does not perform allocation at initialization and defaulting to Eigen behavior causes this issue. More info [here](https://eigen.tuxfamily.org/dox/group__TutorialMatrixClass.html).

## Steps to reproduce

1. Create a sample test.
2. Initialize `Point3()` with the default constructor.
3. Inspect the values. They should be all NaNs.

## Expected behavior

The default should be all zeros. So `Point3()` should give `[0.0, 0.0, 0.0]`. This is currently achievable by doing `Point3::identity()`.

## Environment

Ubuntu 16.04
GTSAM4 with the included Eigen.

## Additional information

varunagrawal, 2019-11-08:
@dellaert I can fix this super quick if you agree this should be fixed quickly. The major issue is a lot of people still having the `Point3()` mindset from GTSAM 3 and then having their results break without any warning or runtime error.

dellaert, 2019-11-08:
Varun, this is desired behavior. There is a flag in GTSAM now that you can specify point3 to be an actual eigen Vector3, rather than its own class. in that case there is no constructor except the Eigen  constructor which does not initialize. Whatever people are used to, in v4 for it is a bug to assume there is a zero initialization.

closed 2019-11-08 by dellaert

ProfFan, 2019-11-08:
Hmm. Just block the default constructor and we are good 2 go?

dellaert, 2019-11-08:
Access to the default constructor is desired. And it is used in many different projects/companies, in the correct way, i.e. with Eigen behavior. This is not a matter of changing the code, this is a matter of documentation.

varunagrawal, 2019-11-08:
But how is this good behavior? Is having NaN initialization (without warning) ever good behavior? Then what becomes the purpose of wrapping Point3 around Vector3d?

ProfFan, 2019-11-08:
I am more like neutral on this, however I think at least we should have a warning for the default constructor in Debug builds.

dellaert, 2019-11-08:
It is the middle of the night in China :-) let’s discuss in detail when I get back. A warning in the debug is also not a solution, because it is *desired* behavior. Let us close this issue and talk in person.