borglab/gtsam #162: what():  Attempting to at the key "0", which does not exist in the Values

hunkyu, 2019-11-09:
## Description

## Steps to reproduce
1.terminate called after throwing an instance of 'gtsam::ValuesKeyDoesNotExist'
  what():  Attempting to at the key "0", which does not exist in the Values.

## Expected behavior
`    NonlinearFactorGraph gtSAMgraph_;
    Values initial_estimate_;
    ISAM2 *isam;
    Values isam_current_estimate_;

#define POSE_SYMBOL ('p')
#define GPS_COORD_SYMBOL ('g')
//#define CALIB_SYMBOL ('c')

#define POSE_KEY(index) (gtsam::Symbol(POSE_SYMBOL, (index)))

    gtsam::ISAM2Params parameters;
    parameters.relinearizeThreshold = 0.01;
    parameters.relinearizeSkip = 1;
    gtsam::ISAM2DoglegParams dogleg_param;
    dogleg_param.setVerbose(false);
    parameters.setOptimizationParams(dogleg_param);
    isam = new ISAM2(parameters);

gtSAMgraph_.add(PriorFactor<Pose3>(POSE_KEY(0), Pose3(Rot3::RzRyRx(roll, pitch, yaw),
initial_estimate_.insert(POSE_KEY(0), Pose3(Rot3::Quaternion(qt.w(), qt.x(), qt.y(), qt.z()), Point3(_current_pose.x, _current_pose.y, _current_pose.z)));
    isam->update(gtSAMgraph_,initial_estimate_);
    isam->update();
    gtSAMgraph_.resize(0);
    initial_estimate_.clear();

`

## Environment

## Additional information

varunagrawal, 2019-11-10:
Are you sure your code is syntactically correct? This example would not compile.

hunkyu, 2019-11-10:
@varunagrawal Can be compiled, is the error when running  
If you don't use Symbol, the operation is normal, like this:

gtSAMgraph_.add(PriorFactor(0, Pose3(Rot3::RzRyRx(roll, pitch, yaw),
initial_estimate_.insert(0, Pose3(Rot3::Quaternion(qt.w(), qt.x(), qt.y(), qt.z()), Point3(current_pose.x, current_pose.y, current_pose.z)));
isam->update(gtSAMgraph,initial_estimate);
isam->update();
gtSAMgraph.resize(0);
initial_estimate_.clear();

dellaert, 2019-11-10:
I agree with Varun that it is hard to see whether this code is really a bug from what we have above. I would recommend adding a `testNonlinearISAM2.cpp` in the tests directory (amazing that we don’t have that yet!), and doing a small pull request with just that. Travis will make sure it’s compiled, and will give a failure if there is indeed a bug.

hunkyu, 2019-11-10:
@dellaert Ok, I am trying

hunkyu, 2019-11-10:
`int main(int argc, char** argv)
{
    NonlinearFactorGraph graph;
    Values initialEstimate;

    ISAM2Params parameters;
    parameters.relinearizeThreshold = 0.01;
    parameters.relinearizeSkip = 1;
    ISAM2 isam(parameters);
    noiseModel::Diagonal::shared_ptr prior_noise_;
    gtsam::Vector vector6(6);
    vector6 << 1e-6, 1e-6, 1e-6, 1e-8, 1e-8, 1e-6;
    prior_noise_ = noiseModel::Diagonal::Variances(vector6);

    int index=0;

    initialEstimate.insert(Symbol('x', index), Pose3(Rot3::RzRyRx(0, 0, 0), Point3(1,1, 1) ));
//    initialEstimate.insert(index, Pose3(Rot3::RzRyRx(0, 0, 0), Point3(1,1, 1) ));
    initialEstimate.print("\nInitial Values:\n");
    graph.add(PriorFactor<Pose3>(Symbol('x', index), Pose3(Rot3::RzRyRx(0, 0, 0),Point3(1, 1, 1)), prior_noise_));
//    graph.add(PriorFactor<Pose3>(index, Pose3(Rot3::RzRyRx(0, 0, 0),Point3(1, 1, 1)), prior_noise_));

    graph.print("\nFactor Graph:\n");

    isam.update(graph,initialEstimate);
    isam.update();
    graph.resize(0);
    initialEstimate.clear();

    Pose3 latest_estimate;
    Values isam_current_estimate_ = isam.calculateEstimate();
    latest_estimate = isam_current_estimate_.at<Pose3>(isam_current_estimate_.size() - 1);
    std::cout << "latest_estimate : " << latest_estimate << std::endl;

    return 0;
}`
-------------------------------------------------------------------------------------
Starting /home/hunkyu/xs/UGVAuto20190409_bus/bin/gtsam_test...

Initial Values:
Values with 1 values:
Value x0: (N5gtsam5Pose3E) R:
[
 	 1  0  0;
    	 0  1  0;
    	-0  0  1
  ]
[1, 1, 1]';

Factor Graph:
size: 1

Factor 0: PriorFactor on x0
  prior mean: R:
[
 	 1  0  0;
    	 0  1  0;
    	-0  0  1
  ]
[1, 1, 1]';  noise model: diagonal sigmas[0.001; 0.001; 0.001; 0.0001; 0.0001; 0.001];

terminate called after throwing an instance of 'gtsam::ValuesKeyDoesNotExist'
  what():  Attempting to at the key "0", which does not exist in the Values.

hunkyu, 2019-11-10:
int main(int argc, char** argv)
{
    NonlinearFactorGraph graph;
    Values initialEstimate;
    ISAM2Params parameters;
    parameters.relinearizeThreshold = 0.01;
    parameters.relinearizeSkip = 1;
    ISAM2 isam(parameters);
    noiseModel::Diagonal::shared_ptr prior_noise_;
    gtsam::Vector vector6(6);
    vector6 << 1e-6, 1e-6, 1e-6, 1e-8, 1e-8, 1e-6;
    prior_noise_ = noiseModel::Diagonal::Variances(vector6);

    int index=0;

//    initialEstimate.insert(Symbol('x', index), Pose3(Rot3::RzRyRx(0, 0, 0), Point3(1,1, 1) ));
    initialEstimate.insert(index, Pose3(Rot3::RzRyRx(0, 0, 0), Point3(1,1, 1) ));
    initialEstimate.print("\nInitial Values:\n");
//    graph.add(PriorFactor<Pose3>(Symbol('x', index), Pose3(Rot3::RzRyRx(0, 0, 0),Point3(1, 1, 1)), prior_noise_));
    graph.add(PriorFactor<Pose3>(index, Pose3(Rot3::RzRyRx(0, 0, 0),Point3(1, 1, 1)), prior_noise_));

    graph.print("\nFactor Graph:\n");

    isam.update(graph,initialEstimate);
    isam.update();
    graph.resize(0);
    initialEstimate.clear();

    Pose3 latest_estimate;
    Values isam_current_estimate_ = isam.calculateEstimate();
    latest_estimate = isam_current_estimate_.at<Pose3>(isam_current_estimate_.size() - 1);
    std::cout << "latest_estimate : " << latest_estimate << std::endl;

    return 0;
}
--------------------------------------------------------------------------------------
Starting /home/hunkyu/xs/UGVAuto20190409_bus/bin/gtsam_test...

Initial Values:
Values with 1 values:
Value 0: (N5gtsam5Pose3E) R:
[
 	 1  0  0;
    	 0  1  0;
    	-0  0  1
  ]
[1, 1, 1]';

Factor Graph:
size: 1

Factor 0: PriorFactor on 0
  prior mean: R:
[
 	 1  0  0;
    	 0  1  0;
    	-0  0  1
  ]
[1, 1, 1]';  noise model: diagonal sigmas[0.001; 0.001; 0.001; 0.0001; 0.0001; 0.001];

latest_estimate : 
|1, 0, 0|
|0, 1, 0|
|0, 0, 1|

[1, 1, 1]';

/home/hunkyu/xs/UGVAuto20190409_bus/bin/gtsam_test exited with code 0

hunkyu, 2019-11-10:
@dellaert

hunkyu, 2019-11-10:
@dellaert  Different in two sentences, Symbol('x', index) and  index

// initialEstimate.insert(Symbol('x', index), Pose3(Rot3::RzRyRx(0, 0, 0), Point3(1,1, 1) ));
initialEstimate.insert(index, Pose3(Rot3::RzRyRx(0, 0, 0), Point3(1,1, 1) ));

// graph.add(PriorFactor(Symbol('x', index), Pose3(Rot3::RzRyRx(0, 0, 0),Point3(1, 1, 1)), prior_noise_));
graph.add(PriorFactor(index, Pose3(Rot3::RzRyRx(0, 0, 0),Point3(1, 1, 1)), prior_noise_));

dellaert, 2019-11-10:
Yeah, I see... ok, we’ll definitely look into it. A PR As I outlined would help. the above is not a test yet.

hunkyu, 2019-11-10:
@dellaert Still I have a problem, I have solved it.
change into 
latest_estimate = isam_current_estimate_.at<Pose3>(Symbol('x', isam_current_estimate_.size() - 1));

closed 2019-11-10 by hunkyu

dellaert, 2019-11-10:
Good find! I definitely did not see it. Glad to know there is no bug :-)

TheSeanParker, 2024-06-21:
①typedef Eigen::Matrix<double,-1,1> VectorXd
②typedef Matrix< Scalar,3,1> Vector3
this bug reason  I find① isnot compatible with②,so will error
https://zhuanlan.zhihu.com/p/704661771