borglab/gtsam #165: Inequality Support in MATLAB/Python

gchenfc, 2019-11-12:
## Feature

Inequality factors and graphs support in MATLAB/Python wrappers

## Motivation

Inequalities are already supported in C++, it would be great to also be supported in the wrappers.

## Pitch

Many problems require inequalities such as joint limits, cable robot tensions, and many other kinematics/dynamics problems.  Bringing inequalities to python would broaden the audience and useability of GTSAM for kinematics, dynamics, motion planning, etc problems.

## Alternatives

Custom nonlinear factors, i.e. barrier function

## Additional context

octopuscabbage, 2019-11-15:
Could this be used for limiting velocity? I have thought that this might improve imu based navigation by allowing you to specify a max velocity that your robot could achieve.

gchenfc, 2019-11-15:
Yes it could!

varunagrawal, 2020-03-30:
`LinearInequality` and `InequalityFactorGraph` are currently under `gtsam_unstable`. It would be great to have them working to a point where we can migrate them to the `gtsam` namespace. Wrapping would then be a cinch.

In the meantime, wrapping via `gtsam_unstable.h` should be easily doable.

ProfFan, 2021-07-19:
@gchenfc Friendly ping :)