borglab/gtsam #177: Use isam to optimize in real time, then use gtsam :: writeG2o to get the results

hunkyu, 2019-12-06:
## Description

## Steps to reproduce

1.     this->isam->update(*this->graph_,*this->initial_estimate_);
        this->isam->update();
        this->graph_->resize(0);
        this->initial_estimate_->clear();
     this->isam_current_estimate_ = this->isam->calculateEstimate();
2.   Finally use gtsam :: writeG2o(*graph_,isam_current_estimate_,"map.g2o") ;

## Expected behavior
3. Found that this saved g2o file only has VERTEX_SE3: QUAT and no EDGE_SE3: QUAT

## Environment
ubuntu 16.04 

I tried to make graph_ do not resize and perform gtsam :: writeG2o, but I got a segmentation fault when saving the g2o file
## Additional information

dellaert, 2019-12-06:
Thanks. Did you look at the `writeG2o` code to see why that could be the case?

hunkyu, 2019-12-07:
Yes, I checked the source code, there is EDGE_SE3: QUAT saved in the code, but the for loop did not enter, because graph-> size = 0, there is no way to save it.

hunkyu, 2019-12-07:
@dellaert  I feel that this writeG2o function can only be used for batch optimization, not suitable for isam optimization.

dellaert, 2019-12-07:
I recommend going through the examples of iSAM and the header files to get a better sense of which graphs are involved in what the graph means in the update method. If you still think there is a bug somewhere, please feel free to re-open this issue.

closed 2019-12-07 by dellaert

gogojjh, 2022-05-08:
isam always stores the nonlinear factors
@hunkyu can this code to save g2o:
gtsam::writeG2o(this->isam->getFactorsUnsafe(), isam_current_estimate_, "map.g2o");