borglab/gtsam #178: Segment fault when calling optimization

LiShuaixin, 2019-12-06:
Hi, there. I tried to use isam to solve a non-linear optimization problem, but failed and got a segment fault. I debugged the code and found the error was thrown out once I called `ISAM2::update()`.

Here is part of my code:
header file:
```
class A
{
  ......
protected:
  gtsam::Values initial_;
  gtsam::Values result_;
  gtsam::NonlinearFactorGraph graph_;
  ISAM2 *isam_;
  ......
}
```

implementation file:
```
A::A() { 
  ......
  ISAM2Params params;

  params.relinearizeThreshold = 0.01;
  params.relinearizeSkip = 1;
  
  isam_ = new ISAM2(params);
  ......
}

bool A::update() 
{
  isam_->update(graph_, initial_);
  isam_->update();

  result_ = isam_->calculateEstimate();
  return true;
}
```

I used gdb to debug the code and it showed:
```
(gdb) bt
#0  0x00007ffff7bafd04 in std::_Rb_tree<unsigned long, std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus>, std::_Select1st<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> >, std::less<unsigned long>, tbb::tbb_allocator<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> > >::_M_erase (__x=0xe9fff7ad14e801ce, this=0x7fffffffc168)
    at /usr/include/c++/5/bits/stl_tree.h:1610
#1  std::_Rb_tree<unsigned long, std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus>, std::_Select1st<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> >, std::less<unsigned long>, tbb::tbb_allocator<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> > >::_M_erase (this=0x7fffffffc168, __x=0x7ffff4cb98fd <std::ostream::put(char)+205>)
    at /usr/include/c++/5/bits/stl_tree.h:1612
#2  0x00007ffff7baff1b in std::_Rb_tree<unsigned long, std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus>, std::_Select1st<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> >, std::less<unsigned long>, tbb::tbb_allocator<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> > >::~_Rb_tree (this=0x7fffffffc168, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/stl_tree.h:858
#3  std::map<unsigned long, gtsam::ISAM2Result::DetailedResults::VariableStatus, std::less<unsigned long>, tbb::tbb_allocator<std::pair<unsigned long const, gtsam::ISAM2Result::DetailedResults::VariableStatus> > >::~map (this=0x7fffffffc168, __in_chrg=<optimized out>)
    at /usr/include/c++/5/bits/stl_map.h:96
#4  gtsam::FastMap<unsigned long, gtsam::ISAM2Result::DetailedResults::VariableStatus>::~FastMap (this=0x7fffffffc168, __in_chrg=<optimized out>)
    at /usr/local/include/gtsam/base/FastMap.h:37
#5  gtsam::ISAM2Result::DetailedResults::~DetailedResults (this=0x7fffffffc168, __in_chrg=<optimized out>)
    at /usr/local/include/gtsam/nonlinear/ISAM2Result.h:117
#6  boost::optional_detail::optional_base<gtsam::ISAM2Result::DetailedResults>::destroy_impl (this=0x7fffffffc160)
    at /usr/include/boost/optional/optional.hpp:735
#7  boost::optional_detail::optional_base<gtsam::ISAM2Result::DetailedResults>::destroy (this=0x7fffffffc160)
    at /usr/include/boost/optional/optional.hpp:697
---Type <return> to continue, or q <return> to quit---
#8  boost::optional_detail::optional_base<gtsam::ISAM2Result::DetailedResults>::~optional_base (this=0x7fffffffc160, __in_chrg=<optimized out>)
    at /usr/include/boost/optional/optional.hpp:331
#9  boost::optional<gtsam::ISAM2Result::DetailedResults>::~optional (this=0x7fffffffc160, __in_chrg=<optimized out>)
    at /usr/include/boost/optional/optional.hpp:867
#10 gtsam::ISAM2Result::~ISAM2Result (this=0x7fffffffc060, __in_chrg=<optimized out>) at /usr/local/include/gtsam/nonlinear/ISAM2Result.h:41
```

My environment is ubuntu 14.04 with gtsam 4.0.0. And I also tested it with the latest gtsam-develop. Could anyone please give me some advices? Many thanks!!

dellaert, 2019-12-06:
First, try with debug and no TBB...

It's destructing `gtsam::ISAM2Result` which I do not see in your code.

Gut feeling: not a GTSAM issue, but something else in your code. Make sure to heed linter errors and perhaps run https://www.cprogramming.com/debugging/valgrind.html.

LiShuaixin, 2019-12-07:
Thank you for your reply. I followed your suggestion recompiled gtsam without TBB. However, the issue was still there. Both backtrace and valgrind point the issue to the calling of isam optimization. I'm not sure whether it is a GTSAM issure or erroneous writing in my code. I referred to the example code in GTSAM example to calling `ISAM2::update()` and didn't find the destructing of `gtsam::ISAN2Result`

In example code  `ISAM2_SmartFactorStereo_IMU.cpp`:
```
     ......
      graph.add(imuFactor);

      isam.update(graph, initialEstimate);

      Values currentEstimate = isam.calculateEstimate();

      imu.propState = imu.preintegrated->predict(imu.prevState, imu.prevBias);
      imu.prevState = NavState(currentEstimate.at<Pose3>(X(lastFrame)),
                               currentEstimate.at<Vector3>(V(lastFrame)));
      imu.prevBias = currentEstimate.at<imuBias::ConstantBias>(B(lastFrame));
      imu.preintegrated->resetIntegrationAndSetBias(imu.prevBias);

      graph.resize(0);
      initialEstimate.clear();
      ......
```

Did you mean that I should build a 'gtsam::ISAM2Result' object and destruct it after calling the `ISAM2::update()`?

haotiangu1993, 2020-02-12:
hi this is Haotian. the segmentation fault will occur when i call isam.update(graph,initialEstimate). do you solve your problems by now? if you do. so how can you solve your problem?

LiShuaixin, 2020-02-22:
@haotiangu1993 Hi, I'm not sure why it crashes when calling `update()`, but it can be solved by changing the version of gtsam.

haotiangu1993, 2020-02-22:
Thanks for getting back to me. You are right. I changed from GTsam 4.0.0 to
4.0.2. It did work.

acxz, 2020-05-11:
I believe this issue can be closed.

closed 2020-05-11 by dellaert

haotiangu1993, 2020-05-11:
Sure. Thanks for letting me know.

Best,
Haotian