borglab/gtsam #180: Strange behavior in LM optimizer after robust noise model changes.

erik-nelson, 2019-12-06:
## Description

After [this line change](https://github.com/borglab/gtsam/pull/167/commits/92008c9319c56bb2edc3a3de7da091a21191b306#diff-ce7584529aa6a7492e625ef1d024910aL688), I am observing different (expected), and potentially buggy (unexpected) behavior when running simple LM optimizations. For simple small problems, the LM inner loop seems to repeat unnecessarily now. For larger problems, optimization fails entirely.

I will continue to try to root cause this. I agree with the above changes and believe they are correct, so I was surprised to see worse performance when using robust noise models in some cases.

I have tracked this down to the [linearized cost change within LM](https://github.com/borglab/gtsam/blob/develop/gtsam/nonlinear/LevenbergMarquardtOptimizer.cpp#L164) being negative, which was not the case before.

While investigating this issue, I found another location where LM's inner loop can potentially get stuck. [On this line](https://github.com/borglab/gtsam/blob/develop/gtsam/nonlinear/LevenbergMarquardtOptimizer.cpp#L202), when the system's error is exactly zero, we will keep running the LM inner loop unnecessarily, since `0 !< 0`. I am able to fix this issue for cases where we perfectly optimize a system by changing the line to:
```cpp
double minAbsoluteTolerance = std::max(std::numeric_limits<double>::epsilon(), params_.relativeErrorTol * currentState->error);
```

However, this feels like a bandaid fix to a larger problem.

I am wondering if now, since the robust error is being evaluated correctly, we are able to perfectly solve some systems that we couldn't before, which leads to an error value of exactly zero, which provokes this hidden bug in LM.

## Steps to reproduce

The following unit test demonstrates the difference in behavior.
```cpp
TEST(Test, RobustNoiseModelGtsam) {
  // Build a single transform3 value at (identity, {1, 2, 3}).
  ::gtsam::Pose3 value(::gtsam::Rot3(), ::gtsam::Point3({1, 2, 3}));
  ::gtsam::Values values;
  values.insert(0, value);

  // Build a diagonal noise model with huber robust loss.
  ::gtsam::Vector sigmas(6);
  sigmas << 1e-3, 1e-3, 1e-3, 1e-2, 1e-2, 1e-2;
  ::gtsam::SharedNoiseModel noise_model =
      ::gtsam::noiseModel::Diagonal::Sigmas(sigmas);
  noise_model = ::gtsam::noiseModel::Robust::Create(
      ::gtsam::noiseModel::mEstimator::Huber::Create(1.345), noise_model);

  // Build a single transform3 prior at (identity, {4, 5, 6}).
  ::gtsam::Pose3 z(::gtsam::Rot3(), ::gtsam::Point3({4, 5, 6}));
  auto factor = boost::make_shared<::gtsam::PriorFactor<::gtsam::Pose3>>(
      0, z, noise_model);
  ::gtsam::NonlinearFactorGraph factors;
  factors.push_back(factor);

  // Optimize.
  ::gtsam::LevenbergMarquardtParams params;
  params.verbosityLM = ::gtsam::LevenbergMarquardtParams::SUMMARY;
  ::gtsam::LevenbergMarquardtParams::SetCeresDefaults(&params);
  ::gtsam::LevenbergMarquardtOptimizer optimizer(factors, values, params);
  ::gtsam::Values optimized_values = optimizer.optimize();

  auto optimized = dynamic_cast<const ::gtsam::GenericValue<::gtsam::Pose3>&>(
                       optimized_values.at(0))
                       .value();

  EXPECT(::gtsam::assert_equal(
      optimized, ::gtsam::Pose3(::gtsam::Rot3(), ::gtsam::Point3({4, 5, 6}))));
}
```

In this small example case, the unit test passes (we optimize the value to the correct result). However, the LM output seems strange. Prior to the above line change, this test emitted:
```
Initial error: 349.441, values: 1
iter      cost      cost_change    lambda  success iter_time
   0  1.349730e-03    3.49e+02    1.00e-04     1    4.90e-05
   1  1.499600e-12    1.35e-03    3.33e-05     1    4.02e-05
   2  1.848893e-22    1.50e-12    1.11e-05     1    3.78e-05
   3  0.000000e+00    1.85e-22    3.70e-06     1    3.74e-05
   3  0.000000e+00    1.85e-22    5.93e-05     1    3.96e-05
   3  0.000000e+00    1.85e-22    1.90e-03     1    3.82e-05
   3  2.265017e-24    1.83e-22    1.21e-01     1    3.81e-05
   3  1.624432e-22    2.24e-23    1.55e+01     1    3.79e-05
   3  1.848893e-22    0.00e+00    3.98e+03     1    3.80e-05
```

Post change, this test emits:
```
Initial error: 348.989, values: 1
iter      cost      cost_change    lambda  success iter_time
   0  6.748650e-04    3.49e+02    1.00e-04     1    4.52e-05
   1  7.498000e-13    6.75e-04    3.33e-05     1    3.81e-05
   2  9.244464e-23    7.50e-13    1.11e-05     1    3.56e-05
   3  0.000000e+00    9.24e-23    3.70e-06     1    3.53e-05
   3  0.000000e+00    9.24e-23    5.93e-05     1    3.51e-05
   3  0.000000e+00    9.24e-23    1.90e-03     1    3.41e-05
   3  1.132508e-24    9.13e-23    1.21e-01     1    3.36e-05
   3      inf    9.13e-23    1.55e+01     1    3.06e-05
   3      inf    9.13e-23    3.98e+03     1    3.04e-05
   3      inf    9.13e-23    2.04e+06     1    3.11e-05
   3      inf    9.13e-23    2.08e+09     1    3.13e-05
   3      inf    9.13e-23    4.27e+12     1    2.95e-05
   3      inf    9.13e-23    1.75e+16     1    3.09e-05
   3      inf    9.13e-23    1.43e+20     1    3.07e-05
   3      inf    9.13e-23    2.35e+24     1    3.05e-05
   3      inf    9.13e-23    7.69e+28     1    3.10e-05
Warning:  Levenberg-Marquardt giving up because cannot decrease error with maximum lambda
```

In larger test cases, I am able to cause optimization to fail entirely in cases where previously it succeeded (values do not converge). Some minor changes are expected: we should have faster convergence since we are now using the correct residual function. I will edit this issue ticket to include a larger test case that fails to optimize when I can create one.

Note that this behavior happens with almost any factor type, and almost any value type, with almost any noise model type when Huber loss is added. I have not tested with other robust losses so far.

## Expected behavior

LM optimizer inner loop does not get stuck when using a robust noise model.

## Environment

OS:  Ubuntu 16.04
Language: C++

## Additional information

dellaert, 2019-12-06:
Oops, seems we might have merged #167 too fast? @mikebosse any quick ideas?
Also, @erik-nelson we also recently pushed a change by you, right?
Also, possibly probe this: Is it *only* Huber or any other robust models as well? Adding @varunagrawal

erik-nelson, 2019-12-06:
> Oops, seems we might have merged #167 too fast?

Possibly. After taking a look, I think that #167's changes are correct, and maybe they just exposed an existing problem.

> Also, @erik-nelson we also recently pushed a change by you, right?

Yup, that's right. I have verified that if I revert my change in #170  the issue persists, and that I can toggle the issue on/off by reverting [this single line](https://github.com/borglab/gtsam/pull/167/commits/92008c9319c56bb2edc3a3de7da091a21191b306#diff-ce7584529aa6a7492e625ef1d024910aL688).

michaelbosse, 2019-12-07:
I believe the issue might be that the absolute error is so small that the LM `tryLambda()` function no longer detects a significant reduction of the error.  
See https://github.com/borglab/gtsam/blob/823de027add1893b7858d2bf3dba49061bed6260/gtsam/nonlinear/LevenbergMarquardtOptimizer.cpp#L188-L192

perhaps the `linearizedCostChange` should instead be relative to the old error? something like this:
`if ( linearizedCostChange / currentState->error > 1e-12 ) { ...`

erik-nelson, 2019-12-09:
@michaelbosse Yeah I agree, that's what I am seeing too. Do you know why this would happen now though?

We're now computing `NoiseModel::distance` and `NonlinearFactor::error`, afaik, correctly as of your change, so we should expect to see faster convergence, and therefore see this issue exposed. However I am curious why we would not have seen this when optimizing something more simple, like the same factor graph with no robust noise models.

michaelbosse, 2019-12-10:
Francesco (fpapi@zoox.com) and I think we found the problem in the LM code.  It was assuming that the nonlinear error and the linear error at zero delta were the same, which is no longer the case when using robust noise models.
This is because the linearized error is the weighted sum of errors which not equal to the robust noise distance function.
Here is a pr for the fix: https://github.com/borglab/gtsam/pull/187

michaelbosse, 2019-12-10:
I had a quick look at the other optimizers (gauss-newton, dogleg and conjugate gradient) and they all seem to be ok since they don't directly compare nonlinear errors with linear errors.

varunagrawal, 2020-07-07:
@erik-nelson has this issue been resolved? It seems that #187 fixes this, but I would like to get confirmation from you.

varunagrawal, 2020-12-01:
@erik-nelson gentle reminder.

varunagrawal, 2021-03-22:
Closing due to lack of a response. @erik-nelson if this is still an issue, please reopen.

closed 2021-03-22 by varunagrawal