borglab/gtsam #182: testSubgraphPreconditioner bad_alloc exception under i386

jlblancoc, 2019-12-07:
Another spin off of #168 . 

Building for i386, the unit test testSubgraphPreconditioner fails. 
Here's a gdb stack trace: 

```
Thread 1 "testSubgraphPre" hit Catchpoint 1 (exception thrown), 0xb7a0d426 in __cxa_throw () from /usr/lib/i386-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0xb7a0d426 in __cxa_throw () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#1  0x0044eaf6 in Eigen::internal::throw_std_bad_alloc() ()
#2  0xb7e56d70 in gtsam::getSubvector(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, gtsam::KeyInfo const&, std::vector<unsigned long long, tbb::tbb_allocator<unsigned long long> > const&) ()
   from /home/jlblanco/code/gtsam/build/gtsam/libgtsam.so.4
#3  0xb7e59c22 in gtsam::SubgraphPreconditioner::solve(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, Eigen::Matrix<double, -1, 1, 0, -1, 1>&) const ()
   from /home/jlblanco/code/gtsam/build/gtsam/libgtsam.so.4
#4  0x004432f8 in SubgraphSolverSolvesTest::run(TestResult&) ()
#5  0x0046ee4c in TestRegistry::run(TestResult&) ()
#6  0x0046f45d in TestRegistry::runAllTests(TestResult&) ()
#7  0x004398da in main ()
```

Will have to re-run with debug info to investigate what's wrong...

ProfFan, 2019-12-22:
I tested on a i386 box on DigitalOcean:

```
 33/240 Test  #33: testPose2 ..............................***Exception: SegFault  0.12 sec

 41/240 Test  #41: testPose3 ..............................***Failed    0.01 sec
not equal:
expected = [
 	       1        0        0        0 0.160001 0.119999;
    	       0     -0.8     -0.6      0.2        0        0
  ]
actual = [
 	   1    0    0   -0 0.16 0.12;
    	  -0 -0.8 -0.6  0.2    0    0
  ]
actual - expected = [
 	-7.77686e-08            0            0           -0 -1.17911e-06  7.38653e-07;
    	          -0  1.66678e-07  3.70296e-08  4.44531e-07            0            0
  ]
/root/gtsam/gtsam/geometry/tests/testPose3.cpp:663: Failure: "assert_equal(expectedH1, actualH1)"
not equal:
expected = [
 	        0 -0.160001 -0.119999;
    	     -0.2         0         0
  ]
actual = [
 	    0 -0.16 -0.12;
    	 -0.2     0     0
  ]
actual - expected = [
 	           0  1.17898e-06 -7.38653e-07;
    	-4.44531e-07            0            0
  ]
/root/gtsam/gtsam/geometry/tests/testPose3.cpp:664: Failure: "assert_equal(expectedH2, actualH2)"
There were 2 failures

        Start  42: testOrientedPlane3
 42/240 Test  #42: testOrientedPlane3 .....................***Failed    0.02 sec
not equal:
expected = [
 	   0.993351 -0.00355884           0;
    	-0.00309235    0.998336           0;
    	          0           0           1
  ]
actual = [
 	   0.993351 -0.00355884           0;
    	-0.00309235    0.998337           0;
    	          0           0           1
  ]
actual - expected = [
 	-3.34242e-07 -1.00049e-09            0;
    	 1.04166e-09  2.80199e-07            0;
    	           0            0 -6.55098e-12
  ]
/root/gtsam/gtsam/geometry/tests/testOrientedPlane3.cpp:174: Failure: "assert_equal(H_expected_numerical, H_actual, 1e-9)"
not equal:
expected = [
 	           1  1.63311e-12            0;
    	-4.51534e-13            1            0;
    	           0            0            1
  ]
actual = [
 	          1 4.92223e-18           0;
    	2.42935e-19           1           0;
    	          0           0           1
  ]
actual - expected = [
 	-2.25501e-07 -1.63311e-12            0;
    	 4.51535e-13 -1.98395e-07            0;
    	           0            0  3.78578e-11
  ]
/root/gtsam/gtsam/geometry/tests/testOrientedPlane3.cpp:180: Failure: "assert_equal(H_expected_numerical, H_actual, 1e-9)"
There were 2 failures

 48/240 Test  #48: testUnit3 ..............................***Exception: SegFault  0.13 sec
not equal:
expected = [
 	  0.998321  -0.013266;
    	0.00666626   0.993267
  ]
actual = [
 	  0.998321  -0.013266;
    	0.00666625   0.993267
  ]
actual - expected = [
 	-4.84334e-07  2.26646e-09;
    	-3.23402e-09 -1.69693e-07
  ]
/root/gtsam/gtsam/geometry/tests/testUnit3.cpp:384: Failure: "assert_equal(H_expected_numerical, H, 1e-9)"
not equal:
expected = [
 	-1.53954e-12     0.222374      0.14825;
    	   -0.257541   -0.0396212    0.0594323
  ]
actual = [
 	-6.77626e-18     0.222375      0.14825;
    	   -0.257539   -0.0396214    0.0594322
  ]
actual - expected = [
 	 1.53953e-12  1.15186e-06 -2.69315e-07;
    	 1.23771e-06 -2.05232e-07 -1.07966e-07
  ]

54/240 Test  #54: testPoint2 .............................***Exception: SegFault  0.12 sec

95/240 Test  #95: testRegularHessianFactor ...............***Exception: SegFault  0.13 sec

        Start 131: testGeneralSFMFactor_Cal3Bundler
131/240 Test #131: testGeneralSFMFactor_Cal3Bundler .......***Exception: SegFault  0.13 sec

        Start 132: testRotateFactor
132/240 Test #132: testRotateFactor .......................   Passed    0.01 sec
        Start 133: testSlamExpressions
133/240 Test #133: testSlamExpressions ....................   Passed    0.01 sec
        Start 134: testGeneralSFMFactor
134/240 Test #134: testGeneralSFMFactor ...................***Exception: SegFault  0.13 sec
149/240 Test #149: testGraph ..............................***Exception: SegFault  0.11 sec

159/240 Test #159: testPCGSolver ..........................***Failed    5.61 sec
/root/gtsam/tests/testPCGSolver.cpp:129: Failure: "Exception: Requested variable 'x1' is not in this VectorValues."
/root/gtsam/tests/testPCGSolver.cpp:150: Failure: "Exception: std::bad_array_new_length"
/root/gtsam/tests/testPCGSolver.cpp:171: Failure: "Exception: Requested variable 'x1' is not in this VectorValues."
There were 3 failures
162/240 Test #162: testSubgraphPreconditioner .............***Exception: SegFault  0.12 sec

185/240 Test #185: testSimPolygon2D .......................***Exception: SegFault  0.13 sec

202/240 Test #202: testIMUSystem ..........................***Exception: SegFault  0.13 sec

218/239 Test #218: testTSAMFactors ........................***Exception: SegFault  0.12 sec
        Start 219: testSmartRangeFactor
219/239 Test #219: testSmartRangeFactor ...................***Exception: SegFault  0.13 sec

```

```
The following tests FAILED:
	 33 - testPose2 (SEGFAULT)
	 41 - testPose3 (Failed)
	 42 - testOrientedPlane3 (Failed)
	 48 - testUnit3 (SEGFAULT)
	 54 - testPoint2 (SEGFAULT)
	 95 - testRegularHessianFactor (SEGFAULT)
	131 - testGeneralSFMFactor_Cal3Bundler (SEGFAULT)
	134 - testGeneralSFMFactor (SEGFAULT)
	149 - testGraph (SEGFAULT)
	159 - testPCGSolver (Failed)
	162 - testSubgraphPreconditioner (SEGFAULT)
	185 - testSimPolygon2D (SEGFAULT)
	202 - testIMUSystem (SEGFAULT)
	218 - testTSAMFactors (SEGFAULT)
	219 - testSmartRangeFactor (SEGFAULT)
```
One test hangs: `testConcurrentIncrementalSmootherDL`.

ProfFan, 2019-12-23:
```
root@ubuntu-c-2-4gib-sfo2-01:~/gtsam_build# ctest -C RelWithDebInfo --output-on-failure -R testPCGSolver
Test project /root/gtsam_build
    Start 158: testPCGSolver
1/1 Test #158: testPCGSolver ....................***Failed    0.90 sec
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::serialization::extended_type_info_typeid<gtsam::Subgraph>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::Subgraph>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::serialization::extended_type_info_typeid<std::vector<gtsam::Subgraph::Edge, std::allocator<gtsam::Subgraph::Edge> > >'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::archive::detail::oserializer<boost::archive::text_oarchive, std::vector<gtsam::Subgraph::Edge, std::allocator<gtsam::Subgraph::Edge> > >'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::serialization::extended_type_info_typeid<gtsam::Subgraph::Edge>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::archive::detail::oserializer<boost::archive::text_oarchive, gtsam::Subgraph::Edge>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::archive::detail::iserializer<boost::archive::text_iarchive, gtsam::Subgraph>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::archive::detail::iserializer<boost::archive::text_iarchive, std::vector<gtsam::Subgraph::Edge, std::allocator<gtsam::Subgraph::Edge> > >'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
/usr/include/boost/serialization/singleton.hpp:132:13: runtime error: reference binding to null pointer of type 'const boost::archive::detail::iserializer<boost::archive::text_iarchive, gtsam::Subgraph::Edge>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /usr/include/boost/serialization/singleton.hpp:132:13 in
=================================================================
==1400==ERROR: AddressSanitizer: stack-buffer-overflow on address 0xbfe9ccac at pc 0xb38d13b4 bp 0xbfe9cc68 sp 0xbfe9cc60
READ of size 4 at 0xbfe9ccac thread T0
    #0 0xb38d13b3 in gtsam::KeyInfo::initialize(gtsam::GaussianFactorGraph const&) /root/gtsam/gtsam/linear/IterativeSolver.cpp:119:43
    #1 0xb38d150b in gtsam::KeyInfo::KeyInfo(gtsam::GaussianFactorGraph const&) /root/gtsam/gtsam/linear/IterativeSolver.cpp:108:3
    #2 0xb38cf9e7 in gtsam::IterativeSolver::optimize(gtsam::GaussianFactorGraph const&, boost::optional<gtsam::KeyInfo const&>, boost::optional<std::map<unsigned long long, Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::less<unsigned long long>, std::allocator<std::pair<unsigned long long const, Eigen::Matrix<double, -1, 1, 0, -1, 1> > > > const&>) /root/gtsam/gtsam/linear/IterativeSolver.cpp:89:45
    #3 0xb3c90b5a in gtsam::NonlinearOptimizer::solve(gtsam::GaussianFactorGraph const&, gtsam::NonlinearOptimizerParams const&) const /root/gtsam/gtsam/nonlinear/NonlinearOptimizer.cpp:154:31
    #4 0xb3dcaeb4 in gtsam::LevenbergMarquardtOptimizer::tryLambda(gtsam::GaussianFactorGraph const&, gtsam::VectorValues const&) /root/gtsam/gtsam/nonlinear/LevenbergMarquardtOptimizer.cpp:149:13
    #5 0xb3dd0eb5 in gtsam::LevenbergMarquardtOptimizer::iterate() /root/gtsam/gtsam/nonlinear/LevenbergMarquardtOptimizer.cpp:288:11
    #6 0xb3c8bc04 in gtsam::NonlinearOptimizer::defaultOptimize() /root/gtsam/gtsam/nonlinear/NonlinearOptimizer.cpp:94:5
    #7 0x815ebba in gtsam::NonlinearOptimizer::optimize() /root/gtsam/gtsam/nonlinear/NonlinearOptimizer.h:98:38
    #8 0x815566a in PCGSolverdummyTest::run(TestResult&) /root/gtsam/tests/testPCGSolver.cpp:143:69
    #9 0x825077d in TestRegistry::run(TestResult&) /root/gtsam/CppUnitLite/TestRegistry.cpp:62:15
    #10 0x824ff65 in TestRegistry::runAllTests(TestResult&) /root/gtsam/CppUnitLite/TestRegistry.cpp:29:22
    #11 0x81582c8 in main /root/gtsam/tests/testPCGSolver.cpp:193:10
    #12 0xb142d636 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18636)
    #13 0x80a09a6 in _start (/root/gtsam_build/tests/testPCGSolver+0x80a09a6)

Address 0xbfe9ccac is located in stack of thread T0 at offset 44 in frame
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==1400==ABORTING
```

ProfFan, 2019-12-23:
`i386`, clang 9.0.1, `RelWithDebInfo`:
```
92% tests passed, 15 tests failed out of 178

Total Test time (real) =  29.45 sec

The following tests FAILED:
	 43 - testTriangulation (Failed)
	 57 - testSerializationGeometry (Failed)
	 69 - testSerializationSymbolic (Failed)
	 87 - testSerializationLinear (Failed)
	105 - testSerializationNonlinear (Failed)
	109 - testRangeFactor (Failed)
	110 - testBearingRangeFactor (Failed)
	111 - testBearingFactor (Failed)
	114 - testSmartProjectionPoseFactor (Failed)
	117 - testSmartFactorBase (Failed)
	123 - testSmartProjectionFactor (Failed)
	140 - testAttitudeFactor (Failed)
	143 - testImuFactorSerialization (Failed)
	158 - testPCGSolver (Failed)
	161 - testSubgraphPreconditioner (Failed)
```

ProfFan, 2019-12-23:
Now everything points to
```
==9749==ERROR: AddressSanitizer: stack-buffer-overflow on address 0xbfe157cc at pc 0xb658912a bp 0xbfe15788 sp 0xbfe15780
READ of size 4 at 0xbfe157cc thread T0
    #0 0xb6589129 in gtsam::KeyInfo::initialize(gtsam::GaussianFactorGraph const&) /root/gtsam/gtsam/linear/IterativeSolver.cpp:121:30
    #1 0xb65894dc in gtsam::KeyInfo::KeyInfo(gtsam::GaussianFactorGraph const&) /root/gtsam/gtsam/linear/IterativeSolver.cpp:108:3
    #2 0x823b6c4 in SubgraphSolverSolvesTest::run(TestResult&) /root/gtsam/tests/testSubgraphPreconditioner.cpp:228:13
    #3 0x83c5a4c in TestRegistry::run(TestResult&) /root/gtsam/CppUnitLite/TestRegistry.cpp:62:15
    #4 0x83c558f in TestRegistry::runAllTests(TestResult&) /root/gtsam/CppUnitLite/TestRegistry.cpp:29:22
    #5 0x824047c in main /root/gtsam/tests/testSubgraphPreconditioner.cpp:314:10
    #6 0xb44fd636 in __libc_start_main /build/glibc-GoSbp4/glibc-2.23/csu/../csu/libc-start.c:291
    #7 0x8180c44 in _start (/root/gtsam_build/tests/testSubgraphPreconditioner+0x8180c44)

Address 0xbfe157cc is located in stack of thread T0 at offset 44 in frame
    #0 0xb6588bff in gtsam::KeyInfo::initialize(gtsam::GaussianFactorGraph const&) /root/gtsam/gtsam/linear/IterativeSolver.cpp:112

  This frame has 7 object(s):
    [16, 40) 'colspec' (line 113) <== Memory access at offset 44 overflows this variable
    [80, 84) 'key' (line 118)
    [96, 100) 'item' (line 119)
    [112, 120) 'ref.tmp' (line 119)
    [144, 148) 'ref.tmp5' (line 120)
    [160, 172) 'ref.tmp13' (line 122)
    [192, 200) 'tmp' (line 122)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /root/gtsam/gtsam/linear/IterativeSolver.cpp:121:30 in gtsam::KeyInfo::initialize(gtsam::GaussianFactorGraph const&)
Shadow bytes around the buggy address:
  0x37fc2aa0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x37fc2ab0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x37fc2ac0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x37fc2ad0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x37fc2ae0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x37fc2af0: 00 00 00 00 f1 f1 00 00 00[f2]f2 f2 f2 f2 04 f2
  0x37fc2b00: 04 f2 f8 f2 f2 f2 f8 f2 f8 f8 f2 f2 f8 f3 f3 f3
  0x37fc2b10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x37fc2b20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x37fc2b30: 00 00 00 00 00 00 00 00 00 00 00 00 f1 f1 00 00
  0x37fc2b40: 00 00 00 00 00 00 00 00 00 00 00 00 f2 f2 f2 f2
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==9749==ABORTING
```

ProfFan, 2019-12-23:
(cont.d) whose reason is that `const map<Key, size_t> colspec = fg.getKeyDimMap();` does not have the key.

This maps to that `fg.getKeyDimMap()` is misbehaving.

```
SIZE: 9
[8646911284551353321, 1001, 2], [8646911284551353322, 1002, 2], [8646911284551353323, 1003, 2], [8646911284551354321, 2001, 2], [8646911284551354322, 2002, 2], [8646911284551354323, 2003, 2], [8646911284551355321, 3001, 2], [8646911284551355322, 3002, 2], [8646911284551355323, 3003, 2],
KEY:1001
1
```

Note that `8646911284551353321` is `0x78*2^{63-8}-1001`, and `(size_t) k` is indeed `1001`.

@jlblancoc @dellaert 
In short, the issue is with the wrong assumption that `size_t` is the same with `Key` (`uint_64` always).

```c++
void KeyInfo::initialize(const GaussianFactorGraph &fg) {
  const map<Key, size_t> colspec = fg.getKeyDimMap();
  const size_t n = ordering_.size();
  size_t start = 0;
  cout << "SIZE: " << colspec.size() << endl;
  for (const auto &[k, v] : colspec)
    cout << "[" << k << ", " << (size_t)k << ", " << v << "], ";
  cout << endl;
  for (size_t i = 0; i < n; ++i) {
    const size_t key = ordering_[i];
    cout << "KEY:" << key << endl;
    auto item = colspec.find(key);
    cout << (item == colspec.end()) << endl;
    const size_t dim = item->second;
    this->emplace(key, KeyInfoEntry(i, dim, start));
    start += dim;
  }
  numCols_ = start;
}
```

dellaert, 2019-12-23:
I see. Well, indeed we want Key to always be 64 bit, so we should track
down we’re this goes wrong....

jlblancoc, 2019-12-23:
Great debugging! :+1: 

I think the line to blame is 118 in IterativeSolver.cpp
```
    const size_t key = ordering_[i];
```
Will open a PR for you to test, @ProfFan .

commit 1b697b19 (2019-12-23): fix Key as size_t (Closes #182)
commit 76ff65f2 (2019-12-23): Merge pull request #199 from borglab/fix/segfaults-in-x32

fix Key as size_t (Closes #182)
PR #183 (2020-05-09): Enable and fix unit tests in different archs
This PR does: 
- [X] Enable running unit tests in Debian builds.
- [X] Relax numerical thresholds in unit tests. Fixes #168 
- [x] Fixes #182 
- [x] Fixes #181 
- [x] Fixes #21 ==> Done in #185 

---
This change is [[image]](https://reviewable.io/reviews/borglab/gtsam/183)
closed 2019-12-23 by dellaert